motioneye-project/motioneye #1093: MJPEG URL Format Is Incorrect

Queuecumber, 2018-12-21:
I have a camera which uses MJPEG for its stream. When I add the camera through motioneye, I get the gray "Unable to open video device" image. Examining the motion debug output shows that it cannot connect to the camera.

This is because thread-1 config file that motioneye has written uses the URL of the camera as `http://whatever/mjpeg` when motioneye expects the URL to be `mjpeg://whatever/mjpeg`. Indeed making this change manually and restarting fixes the problem. The GUI does not allow me to specify an `mjpeg://` url on my own, it says thats an invalid protocol ("protocol mjpeg not supported")

So either I need to be able to specify `mjpeg://` from the GUI or it should write `mjpeg://` itself when it detects that I am using an MJPEG camera (which it does correctly).

ccrisan, 2018-12-26:
@Queuecumber MJPEG is normally transmitted over HTTP, so the URL must begin with `http://`. You don't need to specify the url using `mjpeg://`. The fact that you use `http://` for a network camera implies that it's a MJPEG camera. The other type of supported network camera is RTSP,  which actually starts with `rtsp://`.

Queuecumber, 2018-12-26:
Unfortunately that's not the case. It is correct that mjpeg used http for transport but the motion project config expects an mjpeg:// URL for some reason

Please refer to https://motion-project.github.io/motion_config.html#netcam_url for reference. As I stated in my original issue manually editing the thread config to replace http:// with mjpeg:// fixed my problem 

I encourage you to try this with the latest docker image as an experiment

Queuecumber, 2018-12-29:
I think this is definitely caused the options that Motion provides to ffmpeg.

When I run my stream using ffplay, e.g.

```
ffplay http://10.0.2.10:5000/mjpeg
```

I get lots of errors on the console like 

```
[mpjpeg @ 0x7f991c000b80] Expected boundary '--' not found, instead found a line of 38 bytes
```

because ffmpeg is picking the default encoder as a multipart jpeg which is somehow confused about what the multipart boundary of the images looks like. This is  what using an http:// url with Motion does.

An mjpeg:// url however explicitly passes `-f mjpeg` to ffmpeg. When I do this using ffplay

```
ffplay -f mjpeg http://10.0.2.10:5000/mjpeg
```

I can see the stream without debug errors. Ability to explicitly pass that mjpeg URL is important for working with these cameras for whatever reason.

ccrisan, 2019-01-02:
Closed via #1102.

commit 1189c5c4 (2019-01-02): Allow specifying MJPEG urls

ffmjpeg has issues playing certain mjpeg streams. The mjpeg:// url can be used to work with these streams. Fixes #1093
closed 2019-01-02 by ccrisan