motioneye-project/motioneye #1094: [feature] removing old files on Google drive

bob-lee, 2018-12-23:
Have used `motioneye` for a month now, quite happy and thanks for this project.
Found a similar feature request #200 but I am raising a new one here:

I am preserving movies for 5 days and uploading them to my Google drive.
Old files and folders on local RPI are being removed well (`_remove_older_files` of `mediafiles.py`) but on Google drive, the named folder seems to keep growing and I would have to visit my drive periodically to remove old ones manually (if I am missing something here please advise).

Certainly Google REST API seems to allow removing files and I am interested in contribute to this project   if some guidance is provided (I am web developer, quite new to python):

- how to make code change and test it locally?
- can I just reuse key and tokens in `uploadservices.json`?

bob-lee, 2018-12-23:
Meant to raise this on motioneyeos. Closing..

bob-lee, 2019-01-08:
I am coming back here as I don't know how to make code change and test in `motioneyeos`.

Like to share my approach here:

Assuming that upload service and file persistence are configured and functioning ok, old media folders on local SD card should be deleted by `motioneye`, so I borrowed `uploadservices.py` and some related files and added some codes to:
- get folder list in local/cloud
- delete a cloud folder that does not exist in local

Added codes can be run daily by `crontab` as detailed here:

https://github.com/bob-lee/motioneyecleancloud

```python
    def clean_cloud(self, cloud_dir, local_folders):
        # to delete old cloud folder that does not exist in local
        # assumes 'cloud_dir' is under the 'root'

        count_deleted = 0
        folder_id = self._get_folder_id_by_name('root', cloud_dir, False)
        print 'clean_cloud %s %s' % (cloud_dir, folder_id)
        children = self._get_children(folder_id)
        self.debug('found %s/%s folder(s) in local/cloud' % \
            (len(local_folders), len(children)))
        for child in children:
            id = child['id']
            name = self._get_file_title(id)
            to_delete = not exist_in_local(name, local_folders)
            if to_delete and self._delete_child(folder_id, id):
                count_deleted += 1
                self.info('deleted a folder "%s" on cloud' % name)

        self.info('deleted %s cloud folder(s)' % count_deleted)
        return count_deleted

    def _get_children(self, file_id):
        url = '%s/%s/children' % (self.CREATE_FOLDER_URL, file_id)
        response = self._request(url)

        try:
            response = json.loads(response)

        except Exception:
            self.error("response doesn't seem to be a valid json")
            raise

        return response['items']

    def _delete_child(self, folder_id, child_id):
        url = '%s/%s/children/%s' % (self.CREATE_FOLDER_URL, folder_id, child_id)
        response = self._request(url, None, None, True, 'DELETE')
        succeeded = response == ""
        result = 'success' if succeeded else 'failed'

        return succeeded

    def _get_file_metadata(self, file_id):
        url = '%s/%s' % (self.CREATE_FOLDER_URL, file_id)
        response = self._request(url)

        try:
            response = json.loads(response)

        except Exception:
            self.error("response doesn't seem to be a valid json")
            raise

        return response

    def _get_file_title(self, file_id):
        return self._get_file_metadata(file_id)['title']

def clean_cloud(camera_id, service_name, data):
    camera_config = config.get_camera(camera_id)
    local_dir = camera_config['target_dir']
    cloud_dir = camera_config['@upload_location']

    logging.debug('delete_cloud_folder(%s "%s") %s %s %s' % \
        (camera_id, service_name, settings.CONF_PATH, local_dir, cloud_dir))
    local_folders = get_local_folders(local_dir)
    service = get(camera_id, service_name)
    service.load(data)
    service.clean_cloud(cloud_dir, local_folders)

def exist_in_local(folder, local_folders):
    if not local_folders:
        local_folders = []

    if not folder:
        return False

    return folder in local_folders

def get_local_folders(dir):
    folders = next(os.walk(dir))[1]
    print folders
    return folders
```

JamBalaya56562, 2026-10-26:
This has been implemented by @bob-lee in #1118 (merged in 2019) as the "Clean Cloud" option: with it enabled in the camera's upload settings (Google Drive only), folders removed locally by the "Preserve" cleanup are removed from Google Drive as well. A remaining problem with custom preserve periods is tracked in #1463.

Closing as completed. Feel free to reopen if anything is still missing.

PR #1118 (2019-02-07): clean cloud (gdrive only)
Resolves #1094.
Added an abstract function `UploadService.clean_cloud` and implemented `GoogleDrive.clean_cloud`.
When `mediafiles.py` removes a old local folder, it calls `clean_cloud` to remove old cloud folders as well that do not exist in local.
closed 2026-10-26 by JamBalaya56562