motioneye-project/motioneye #1095: motioneyeos is cooking my raspberry

JBSullivan, 2018-12-24:
motioneyeos cooks my raspberry 🤨🤨🤨🤨🤨

Any ideas why motioneye(os) is heating my raspberry B3+ so much? I use 4 IP Cam and 1 USB Cam on the Raspi. 

The CPU temperature of the Raspi is 68,8°C - the CPU load is 84% - 100%.

1st test
No video streaming, no motion detection, the video resolution is only 640x480, I use the second cam stream if possible. All frame rates are at 2 fps - now the CPU load is about 74% and the CPU temperature is 67.7°C - that's too much again, I think.

2nd test
I erased three out of five cams.  Now the CPU load is 21% and the temperature ~ 50°C.

What is the limit that Motioneye of the webcam can use on the rasbperry B3+? Maybe a raspberry is not the best choice for Motioneye. Are you thinking about sharing a ready-to-install image for a VM (VirtualBox) to use better hardware?

motionEye version 0.39.3
Motion version 4.1.1.1
OS version motionEyeOS 20181209

djraimondi, 2018-12-28:
When using virtual box check your network settings of the VM. If you are using NAT try changing it to "Bridged" network so your VM has direct access to your network adapter and LAN.

JamBalaya56562, 2026-11-08:
Closing as answered by your own tests above: with five cameras the Pi 3B+ ran at 84–100% CPU, and with two it dropped to about 21%. Motion decodes every camera stream on the Pi, so the load grows with the number of cameras, their resolution and frame rate, and a Pi 3 handles only a few. A bigger machine as the hub, with the cameras streaming to it, spreads that load.

closed 2026-11-08 by JamBalaya56562