motioneye-project/motioneye #1096: tinyCam Monitor is able to display stream - how to for motionEye???

olvier, 2018-12-24:
hey folks.

the android app "tinyCam Monitor" is able to show a stream of an "Digoo DG-UFC" cam, which do not have onvif protocol.

The following parameters are exported by tinyCam Monitor.
But i am not able to get it to work in motionEye -.-
Maybe somebody is able to help?

I tried "all" variants of urls, with rtsp://; http://; both ports and so on.

> Parameters used by tinyCam Monitor app to access camera "http://192.168.0.163:80"
>
>
>Name: Digoo DG-UFC 1080p
>Vendor: (P2P)
>Model: TUTK
>Hostname: 192.168.0.163
>Web port: 80
>Protocol: P2P
>RTSP port: 554
>HTTPS: no
>Username: admin
>Password: _(is really empty here, didn't deleted anything)_
>
>-----
 >tinyCam Monitor - complete solution for mobile surveillance for Android
>https://tinycammonitor.com

ccrisan, 2018-12-28:
You'll probably need to add some path at the end of the URL, but that's not something we can help you with. Look on forums and see what others with your camera model have done about it.

closed 2018-12-28 by ccrisan

olvier, 2018-12-28:
Thank you for your fast response. I understand the problem of the addition
to the URL. I also asked the author of tinycam Monitor and I will update
here, if I get some feedback.

Kind regards