motioneye-project/motioneye #1097: Streaming from webui works only after opening streaming port directly

dav01it, 2018-12-25:
I have a netcam on 192.168.0.34
Motioneye on port 8765 with streaming enabled on port 9081.
Rtsp string is correct and I get motion recordings correct but no image in the web interface on port 8765.
If I open the streaming url192.168.0.34:9081 I get the streaming image there and from that moment also on the webui on port 8765.
If I change some settings and the service restarts the problem start again.
Is it a bug or am I missing something?

ccrisan, 2018-12-28:
@dav01it most likely you have added your camera as a Simple MJPEG Camera and your browser isn't capable of displaying the MJPEG.

dav01it, 2018-12-29:
Thank You Calin for Your answer. I tried it with chrome from my phone and also firefox but without success.
It seems that the server itself provides me an image instead of the rtsp stream;
 the code I get is:
`<img class="camera error" style="height: 382px;" src="/picture/1/current/?_=1546109438518&amp;width=509&amp;_username=admin&amp;_signature=4af78dffd8c678eaa6280d972d8a861d1d9fd454">
`
After a deeper analisys I've discovered that to have an updated image I must keep open another tab that points to the streaming url/port: when I close the tab no more updates are provided. So I understand that the main page reads the last uploaded image from streaming port or the occurring motion event: If no motion is detected you will not get images.
I've attached a screenshot of the cfg if it could be useful.
Please tell me if I missed something or if I can help to investigate further.
Thanks again.

[image]

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye 0.39.3, and there were no further replies. As ccrisan said above, this looks like the camera was added as a Simple MJPEG Camera, which only passes the camera's own stream through. For an RTSP camera, add it as a Network Camera, so Motion reads it and the web UI shows its picture. Please open a new issue if that still shows no picture with a current release.

closed 2026-11-08 by JamBalaya56562