motioneye-project/motioneye #1103: Neither dropbox or google drive upload is working

aredon, 2018-12-31:
I deleted uploadservices.json and reconfigured just to be sure. Both services say that the test was successful, and indeed if I give the google drive connection a folder that doesn't exist - it does create the folder in google drive without issue (when testing the connection).

However, neither service will upload videos. motioneye.log is empty and I've no idea why this would be.

ccrisan, 2019-01-09:
Wrong motioneye log path? Try running it meyectl manually with stdout logging.

aredon, 2019-09-02:
I ended up creating my own script and running it through the run a command hooks on file creation. Using a script called Dropbox_uploader.

JamBalaya56562, 2026-11-10:
Closing, as the reporter moved to their own upload script run from the file-created hook, and this was with a 2018 release. Please open a new issue with `motioneye.log` at debug level if uploads still fail with a current motionEye.

closed 2026-11-10 by JamBalaya56562