motioneye-project/motioneye #1105: RPi3 Recorded Video Skipping/Jumping

scha1703, 2019-01-02:
I tried using the search for similar issues but was unsuccessful. 

I'm using an RPi 3 B+ with MotioneyeOS Version 0.39.3 and Motion Version 4.1.1
I've configured motion detection and streaming, all is working well however the saved recordings do not have a smooth playback, they will jump in time at random points. I'm not referring to the FPS because the video is very smooth when it is playing, but it will skip multiple seconds of motion at times. I've viewed the stream while a motion event capture is taking place so I know that it should be recording/saving these parts of the video. It's difficult to explain so I have an example below:

[Video Example](https://drive.google.com/file/d/1JQ0wGNJRUEgYsbtMAF-1FpjHxODWlxe5/view?usp=sharing)

So you can see the FPS is fine, but it completely skips (5 sec mark) over them closing the door (and they actually stood around for a bit before leaving), but the recording just catches them leaving. 

Does this have to deal with the motion parameters? If so could anyone suggest which ones may be the issue? 

Thanks!

jasaw, 2019-01-03:
It looks like your motion threshold and minimum motion frames are not right. First turn on "show debug frames" and it should draw red box around detected motion, and there's number of changed pixels on top right corner too. That should help you tune your detection parameters.

dmishins, 2019-01-04:
I am having the same issue, also with 0.39.3 and 4.1.1.  It is clearly visible by the timestamp in the corner that the video skips about 10 seconds when there is clearly motion.  If the issue was about the motion threshold and minimum motion frames were incorrect, shouldn't it separate the videos into two separate files? Also, I have found situations where the motion in the frames that are skipped is very significant, so that seems unlikely to be the answer.

Thanks,
Daniel

scha1703, 2019-01-04:
@jasaw @dmishins Thank you for the suggestions, I'll keep playing with the settings. My frame change percentage is very small (1.5%) and the minimum motion frames is only set to 5, so I would think it would catch the slightest movements. Is there a possibility the Pi is getting bogged down with vision/motion processing? I'm in the process of changing my setup to a dedicated server (PC running Ubuntu) to handle all vision/motion processing, and just using the Pi as camera 'nodes'.  

Thanks,
Garrett

dmishins, 2019-01-04:
@jasaw Thank you for the help! Are you saying that if the two "movements" are within the motion gap, they will be recorded in the same file but will have a break in the middle of the video?  Also, I am fairly confident that there was motion in the video during the time that was lost. My video is similar to scha's video, where it seems clear that there must have been motion in the dropped frames.  In scha's video, the closing door must have been a change enough to trigger the motion detection.

closed 2019-03-25 by scha1703