motioneye-project/motioneye #1106: Can't reach webinterface after enabling "Fast Network Camera"

JoeTN, 2019-01-02:
I'm running MotioneyeOS on a Raspberry pi model B with 512MB RAM from Q4 2012.
It has been running fine using the CSI camera module.

I just decided to enable the "Fast Network Camera"-option. After rebooting the PI I simply cannot access the webinterface anymore.

motioneye.log states "ERROR: streameye restart failed"
boot.log states " * Starting motioneye: failed"

I can establish SSH access, but only for a couple of minutes before the session is terminated, also I can view the stream at http://192.168.2.38:8081/.

Is there anyway to disable the FNC-option from CLI to regain access to the webinterface ?

Logs:
[streameye.log](https://github.com/ccrisan/motioneye/files/2722192/streameye.log)
[boot.log](https://github.com/ccrisan/motioneye/files/2722193/boot.log)
[motion.log](https://github.com/ccrisan/motioneye/files/2722194/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneye/files/2722195/motioneye.log)

Thanks in advance and best regards
Joe

ccrisan, 2019-01-11:
If you can briefly run an SSH command, you can "reset" stuff there by removing the following files:

    rm -f /data/etc/motioneye.conf /data/etc/raspimjpeg.conf /data/etc/streameye.conf /data/etc/motion.conf /data/etc/thread-*.conf

JoeTN, 2019-01-12:
Thanks for your reply ccrisan, I did what you suggested which did indeed reset the settings, so I have access again now, thanks! - if this is the "official" way of resetting the settings through SSH you should add it to your "wiki" or whatever - I couldn't find the information anywhere ;-)

Also I just re-enabled FNC after gaining access, and I again lost access to the webinterface, is it supposed to work this way ? - I guess there should still be a minimalistic webinterface with some basic settings, or am I wrong on this.

Thanks!

ccrisan, 2019-01-13:
@JoeTN there's no official way of resetting the system. Ideally there would be some button in the UI actually formatting the data partition. This version is a simplified version which doesn't remove your data.

closed 2019-01-13 by ccrisan