motioneye-project/motioneye #1111: After reinstall rtsp capera gives green output

harolds, 2019-01-08:
I had to reinstall my setup because of a bad memory card. So now I used the latest versions of everything :-)

motioneye: 0.39.3
motion: 4.1.1
ffmpeg: 3.2.10

But when I add a network camera (that worked before) I get no image or a green image. Looks like ffmpeg is using hardware encoding/decoding. How do I fix this?

One other strange thing thats happening too is that whatever camera I add the first one is always the USB connected camera. So if I configure a network camera then it shows the USB camera output. Tried to remove everthing and reinstalled it. Does not fix this problem.

wordblind, 2019-01-10:
I have this problem when running UDP over wifi due to dropped packages.
When I changed it to TCP this issue went away, but I had to change the frame rate due to limited bandwidth over WIFI.

JamBalaya56562, 2026-11-08:
Closing, as there were no further replies. A green picture from an RTSP camera usually means Motion's decoder got incomplete frames; as wordblind found above, switching the camera's RTSP transport from UDP to TCP fixed it over Wi-Fi. If a network camera still shows green with current releases, please open a new issue with `motion.log`.

closed 2026-11-08 by JamBalaya56562