motioneye-project/motioneye #1112: Motioneye with Sony PS3 eye camera with microphone - I would like to auto record sound

2boot, 2019-01-09:
This is **not** an issue it is a "like to have".
Possible to skip to last paragraph for problem, other information might help others.
I have a RPi 3B+ fully updated with stretch in November 2018 with two PS3 cameras attached and as far as I can tell everything is working fine, it uploads video recordings by ftp to an online store, it uses Pushover to alert my mobile phone, I have entered debug mode and made reboot : shutdown : backup and restore always available in the admin panel, I can even reboot whilst I am down the town etc etc.. 
**However I would like to record sound at the same time as the video.**

**This is what I can do**
Whilst motioneye is functioning waiting for a detection I can use putty to enter the command 
"arecord -D plughw:1,0 -d 3 test.wav" and record from the microphone. I can then ftp this file to a win10 pc and play it. So far so good it will record manually..

**What I would like to do**
Now what my intention is for a simple guy like me with very little programming experience - is to have the simply named test.wav recording started by some command in the "Run a Command" box under the notification section of motioneye. In future I want to set 58 seconds recording length (not the 3 indicated" to terminate just before the max set video duration of 60 seconds (but better to terminate as video terminates).
I want to save the file in the same local directory as for the video for uploading. The file name needs to be written with %Y-%m-%d/%H-%M-%S to tie up with the video.
In the Run End command I will paste my Pushover python command.

**My initial problem** is that whatever I type into the "Run a Command" box doesn't do anything (even using the "arecord -D plughw:1,0 -d 3 test.wav"  and ignoring the wants/desires ) - I haven't found anywhere **written** that will help me or that I can understand sufficiently. 
Anyone know what to type in the "Run a Command" box to start me off - any other simple tips?
Apologies, but I am struggling - this my first use of github.

jasaw, 2019-01-13:
@2boot If you want better synchronization between video and audio, you could try `extpipe` motion option. Disable video recording in motion, and instead use `extpipe` to record audio video.

Here's what I would try:
1. Install alsa audio driver and audio tools (alsamixer, arecord, ...).
2. Install libfdk AAC encoder. May have to recompile ffmpeg to support libfdk. There are tons of examples on the internet on how to do this. If recompiling ffmpeg, you may want to enable OMX hw video encoder (add `--enable-omx --enable-omx-rpi` configure options).
3. Now it's time to play with motion's extpipe. Add extpipe to thread-1.conf file. Something like this extpipe should encode video and audio into a file: 
`ffmpeg -fflags +nobuffer -re -y -f rawvideo -pix_fmt yuv420p -video_size %wx%h -framerate %fps -i pipe:0 -fflags +nobuffer -re -f alsa -ar 16000 -ac 2 -i hw:1,0 -map 0:0 -map 1:0 -c:v h264_omx -profile:v high -b:v 2000000 -c:a aac -b:a 16k -ac 1 -f  mp4 %f.mp4`. To record movie in mkv format instead of mp4, replace `-f mp4 %f.mp4` with `-f matroska %f.mkv`.
7. Disable motion's pre-capture and motion gap as they will cause havoc to the video stream.

2boot, 2019-01-14:
@jasaw thank you for your show of confidence in my abilities however much misplaced. I also just noticed the previous reply was by your hand - my lack of familiarity with GitHub coming to the surface. Thank you for the pointers I will investigate. I also noticed you gave similar advice in motioneyeos thread #105 which remains open, does this mean so far it has not been done? If so it could be a challenge too far. working...…..

jasaw, 2019-01-15:
@2boot I think you meant this issue in MotionEyeOS: https://github.com/ccrisan/motioneyeos/issues/105

You are right that recording movie with audio has not been implemented yet, and I don't see it happening any time soon. This requires changes to the motion software, so [motion github project](https://github.com/Motion-Project/motion) the best place to request for this feature.

If you still want audio, the alternative is to use extpipe like I suggested. In theory it should work, but haven't tested it.

DeadEnded, 2020-06-24:
As a work around (for anyone who wants to try it) I have worked with a few others to make a script that can add audio to recordings.  I have only tested it on my own setup but am happy to try and help others.

https://github.com/DeadEnded/MotionEyeAudio

2boot, 2020-06-30:
@DeadEnded I am trying to follow your method, but have had failures and successes.

First I reconfigured my Pi3b+ from MotioneyeOS to Motioneye took off the Picam and put the PS3 (all working again, that was the easy bit).

Second I did not know what Docker was, I signed up linked to github, then I found it impossible to install docker in a hyper-v windows 10 pro guest, in the end I installed docker using command line in hyper-v Ubuntu 18.04 guest (installing docker version 19.03.6) and can operate it by CLI (it's just that I don't know what I am aiming to do - I never blindly ask for help unless I have spent many hours trying).

So now for the good news, this is what I have

**root@2boot-Virtual-Machine:/var/lib/docker# docker image ls
REPOSITORY                TAG                 IMAGE ID            CREATED             SIZE
deadend/motioneye-audio   latest              4a6b1f9c97a9        2 weeks ago         411MB
root@2boot-Virtual-Machine:/var/lib/docker#**

So now for the questions
Will (after some docker commands) I end up with a "directory" which I then copy over a "specific directory" by e.g. sftp'ing into my Pi3b+?
I hope I don't have to make alterations by building your work onto the Motioneye source code, as this would be well past my capability.
Assuming the directory approach is correct could you help with docker commands to achieve that directory. I thought I had cracked it when I made a tar file, but the internals of the tar were maybe 8 lines of long long long filenames- so that is where I have paused.

DeadEnded, 2020-06-30:
Hmm... I have not done docker in Windows, so not sure of how the setup would be for that.  If you previously used MotionEyeOS and would like to use that, I have updated (just an hour or so ago) the ReadMe with the directories that MotionEyeOS use.  I don't have this setup, but it should work on there too.

If you want to continue with the Docker install, it looks like (from what you posted) you have downloaded the image.  I don't know if you have spun up a container yet... you can check with `docker container ls`.

If you have not, you can do it with:
```
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    deadend/motioneye-audio:latest
```
The volume mappings (-v) are to the docker host system - I believe you are using Ubuntu VM as the docker host.  So you would need to create directories where you want the config (/etc/motioneye) and saved files (/var/lib/motioneye) on the host.  The first part of the volume map is the host location the second half (after : ) is the container path.  So you would change the first half.

For example if you created a config folder in `/home/motioneye` and you wanted to save the recordings to a drive mounted in  `/ media/motioneye` it would become:
```
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /home/motioneye:/etc/motioneye \
    -v /media/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    deadend/motioneye-audio:latest
```
Hopefully this all makes sense!
Good Luck!
DeadEnd

2boot, 2020-07-01:
I created the two required directories in the Ubuntu VM (as this is the host)

I ran your first command with sudo
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    deadend/motioneye-audio:latest

it completed and when running 'sudo docker container ls' it gives this

2boot@2boot-Virtual-Machine:~$ sudo docker container ls
[sudo] password for 2boot: 
CONTAINER ID        IMAGE                            COMMAND                  CREATED             STATUS              PORTS                    NAMES
78c856882095        deadend/motioneye-audio:latest   "/bin/sh -c 'test -e…"   8 minutes ago       Up 8 minutes        0.0.0.0:8765->8765/tcp   motioneye
2boot@2boot-Virtual-Machine:~$

On checking the contents of the two directories I find

2boot@2boot-Virtual-Machine:~$ ls /etc/motioneye
motioneye.conf
2boot@2boot-Virtual-Machine:~$ ls /var/lib/motioneye
2boot@2boot-Virtual-Machine:~$

I am assuming that this is not as expected, as only motioneye.conf has appeared

**Now for some big questions** 
Am I expected to be connected to my Pi3b+ for the above to work?
Should I have copied some files from the pi3b+ to those locations prior running the command?
Should I have done something else to prepare for the command?
Is the result as expected?

DeadEnded, 2020-07-01:
First, just to clarify, is your VM on the PI or on another computer?
Since the motioneye.conf was created, I think that the container is probably working.
Have you connected to the GUI yet?  Do this by going to the http://192.168.0.101:8765/ using your host IP instead of 192.168.0.101.

The rest of the files that are normally found in the config folder may not appear until you start configuring the GUI.

2boot, 2020-07-01:
@DeadEnded The VM is not on the Pi.
My setup is as follows, I have a windows10pro PC HOST which is running Ubuntu 18.04LTS VM. (Docker on the VM)
My raspberry pi 3b+ is located remotely and could be (if I had done something) connected to over wifi.
In my work so far the raspberry pi3b+ running motioneye has been sitting there waiting for me to do something.
Before working with Docker I could (and hopefully still can) use all of VNC, Filezilla and Putty to get into the pi from my HOST.

I just now **using the VM** and Firefox I entered 192.168.1.23:8765 and I can see the GUI, login as admin, and view the motioneye PS3 camera within a Firefox tab (as I could already from HOST).

I will now consider in detail your last message and that written on Github/Docker concerning names of folders as it seems the intention is to write directly from Docker to the pi3b+.

I think now knowing what the intention is, I may be closer to actually doing it. Sorry to be a bit of a nuisance but I have never done anything like this before. A small concern at the moment comes from the 'sudo docker container ls' command which lists in it's output 0.0.0.0:8765 and not 192.168.1.23:8765. (when you wrote host IP, I am assuming ** _in my case_** it is the VM)
I hope the above is clear.

DeadEnded, 2020-07-01:
It sounds like it is up and running and everything sounds pretty good.  From the docker host (your ubuntu VM) all my containers also show `0.0.0.0:port` as well so you're fine there.  It is just telling you they are the local ports of the host.  So from the sounds of it everything is up and running fine.

The only thing I am not sure about is your comment of `it seems the intention is to write directly from Docker to the pi3b+`.  I'm not sure what you mean here.  So to clarify, the docker volume naming is to map directories from the host (outside the container) to directories inside the container.  The reason for this is that any files in these volumes exist outside the container, and if you delete the container, they are NOT deleted with it.  This allows persistent files.  There is one for config files (`/etc/motioneye/`) and another for saved videos/pictures (`/var/lib/motioneye/`).

As far as the pi3b+, I believe it is just being treated as a camera - MotionEye (your container) is not writing anything to the pi3b+.  Now whatever software you have on the pi3b+ could be recording locally on the pi as well - but all the MotionEye container is doing as far as the camera is consuming the feed.  If you have MotionEyeOS on the pi3b+ it can be configured to save locally to the pi3b+, or it can be setup as a satellite I believe by using one of the settings (fast network camera).

2boot, 2020-07-02:
Hi @deadended your patience is very much appreciated.

From your last paragraph if I understand correctly - videowithsound files will be recorded on the pi3b+ and that if I am using the host with docker (which has now been up for 30hours) those same videowithsound files will also appear in the directories mentioned on the docker host both inside and outside the container whether or not I am firefoxed into the pi3b+ from the docker host.

For the pi3b+ I am using an unmodified 'motioneye' which is installed as per ccrisan procedure from a github repositiory.
I looked inside your motioneye-audio.sh and see reference to ffmpeg. Is ffmpeg already included in an unmodified motioneye? If it is all well and good.

The connected PS3 camera has been selected as (V4L2) (/dev/v4l/by-id/usb-OmniVision_Technologies__Inc._USB_Camera-B4.09.24.1-video-index0). Please state what Movie Format you have selected in the motioneye Movies section that will not inhibit sound capture.

If I wanted to debug the video file, the production of the temp file and rewriting of the final file, would I change motioneye.conf to debug setting on the host or on the pi or should the host auto mirror?

I wrote the above para because presently I have 'reboot true' on the pi3b+ and 'reboot false' in the host file locations and when I firefox in to the pi3b+ **from any browser even pc's without docker** the reboot and shutdown buttons are available.

Just to complicate matters my movie files on the pi will be stored in /var/lib/motioneye/Camera1

Should I now run this
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye**/Camera1** \
    --restart="always" \
    --detach=true \
    deadend/motioneye-audio:latest

If there were two PS3 cameras would I need to add another line
     -v /var/lib/motioneye:/var/lib/motioneye/Camera2 \

or should I change the file store on each camera to be /var/lib/motioneye
and leave the docker command line as
-v /var/lib/motioneye:/var/lib/motioneye \
so that both cameras deposit files in the same pi3b+ directory.

I have made all the changes required - resulting in motioneye-audio.sh green colour in the /etc/motioneye directory on the pi using putty and  placed the extra motion options and run a command options appropriately both using the GUI..

Video files are being produced on the pi but as yet no sound but I want to get the above correct first hence my mention of debug for my testing after your reply, I assume debug will pickup the echo lines in the motioneye-audio.sh file.

Thanks

DeadEnded, 2020-07-02:
The Pi is completely separate from the VM.  I expect the MotionEye Docker container see's the Pi just as a camera itself - so the files saved on the PI are completely separate and not part of the Docker setup.  Think of the Pi as a camera with a stream being fed to the Docker Container.  None of the files on the PI matter - the docker and host are their own system.

So, the directories you are mapping between host and container do not involve the PI.  Also, the saved videos on the PI are also completely separate from what is being saved on the Docker/Host.  The script, and everything involved is only relevant on the docker and host system.

 I hope that clarifies the setup.  The PI is only the stream source - after that everything is on the docker/host system.

2boot, 2020-07-09:
Hi @DeadEnded Your information has hit home thank you for your persistence. I now understand where things are relative to each other, but could I please have a little more help regarding docker as it appears the docker commands are extremely complex.

### STATUS SO FAR
**So I have run your command in a different ubuntu host a laptop ( _reason_ the hyper-v ubuntu VM was flaky and kept disconnecting from the network)**
docker run --name="motioneye"
-p 8765:8765
--hostname="motioneye"
-v /etc/localtime:/etc/localtime:ro
-v /etc/motioneye:/etc/motioneye
-v /var/lib/motioneye:/var/lib/motioneye
--restart="always"
--detach=true
deadend/motioneye-audio:latest

**and the uptime is increasing.**

My Ubuntu host is 192.168.248 so in the host Firefox search bar I typed 192.168.248:8765 (as you previously indicated) and added two "Remote motioneye Cameras"
and both are displaying.

### HELP NEEDED
Could you please supply a command to enter the container to reach what I assume is the container root/file system in order that I can chmod +x  the available motioneye-audio.sh file (when I find it) and a nice way to exit.

===========================
**I think the command starting at root in Ubuntu terminal is** 
`docker run -i deadend/motioneye-audio:latest bash`
followed by
`/bin/sh`

reaching a location without indication where command responses are in single column format

to exit this location
use `exit` 
and again `exit`
to get back to /var/lib/docker

===========================

**I am asking first as I do not want to destroy any of your good work**

The setup for the cameras seen in docker is straight forward I will then try for sound. 
If the PS3 is not successful I will try a logitech device.

When I shutdown the Ubuntu host do I need to do anything to docker **before** I shutdown the laptop?

When I start Ubuntu do I need to run any docker command to get back to where I was prior starting Firefox?

Thanks again for your help

DeadEnded, 2020-07-09:
To enter the container you typically use (assuming your container is named `motioneye`:
```
docker exec -it motioneye /bin/bash
```
That should enter the container to a terminal prompt.
That being said, you shouldn't need to.
That is the benefit of using the volume map - as long as you have the `/etc/motioneye` directory on your host, the container inherits this.  So you just need to go to that directory on your host to make the change.  It will instantly be reflected in the container.  If you don't have that directory on your host - you will probably need to stop the container, create it, and then start the container again.

When you shutdown your laptop I expect the OS will gracefully shutdown the docker container.
When you start up the laptop the docker service should start automatically.  Any containers that are set to auto start will, otherwise you will need to use `docker start motioneye` to start the container.

I hope this is clear and helpful 😃 .
DeadEnd

2boot, 2020-07-11:
@DeadEnded 

Absolutely, 
notes 
I did not find the motioneye-audio.sh in position I had to copy it to the location, 
I have still not tried for sound as I want to get the silent movies correct first.

I believe I have tried every combination of camera settings to add a camera in the container, in some cases I succeed, others include green screen and some just wont etc, but could I please ask for your settings to view a camera on raspberry pi set up with raspian and motioneye, failing that the same settings if you have used motioneyeOS.

The below screen shot shows my tab motioneye  (i.e. inside container) on the host proved by showing OS version Ubuntu 18.10

[image]

Now, should I be adding a "Remote motioneye Camera"? Based upon responses to my previous questions of you I believe you will answer no, **but exactly what does your setup screen show in the boxes Camera Device and Camera Type?** so I can force that first.

Thanks in advance

DeadEnded, 2020-07-12:
I don't use any PI's... just IP cameras.  But looking at your screenshot and comparing it to what I see in videos for MotionEyeOS setup, I think you have the wrong camera device URL.  That URL is for the GUI, not for the camera stream.  From one youtube vid I found, the default initial port is 8081:
[image]

But check your PI setup and make sure streaming is turned on, and then click the `StreamingURL` that you see in the above screen shot.  That should be what you add to MotionEye container to pull up the stream.

This is assuming the PI remote camera isn't different... Again, I don't use them - so I don't know first hand.

Good Luck!

2boot, 2020-07-12:
@DeadEnded You must be dreading when I mention you.
Firstly I didn't mean for you to investigate for me, but with me previously coming against a brick wall and not getting as yet to a satisfactory solution I thought it best to ask for the two config lines.

So OK I understand that you do not use Pi's, even so I write to show what is happening in my case. Please do not take an action except perhaps the last paragraph.
192.168.1.245 is a MotioneyeOS Pizero
I look for it's streaming port as I did before my last message.
[image]
The next image shows the container and two small firefox screens, 
bottom right direct to Pizero from firefox and into motioneye GUI.
top right firefox into streaming port of Pizero.
left motioneye being run in container on Ubuntu host trying to pick up Network camera at http://192.168.1.245:8081 which does not display. 

[image]
It seems that a Pi with a PS3 camera attached is not an  IP camera and neither is a Pi with a Logitech C310 camera attached (whilst both camera devices have video and sound capability).

It seems that no matter what I do I cannot get video recordings into the container (sound not considered as yet).

Could I please ask the manufacturer and the model of your IP camera, I was looking at some this morning which were operated by mobile phone and app although they look good I am not sure what to do next.

DeadEnded, 2020-07-13:
To answer your last paragraph (as request) I currently have both Dahua (a bit expensive) and Wyze Cams ( inexpensive but with hacked firmware) connected to my MotionEye system.  Many, many IP camera's (almost all) should work with MotionEye.  If they support RTSP chances are they will work.

I did find this list of "supported devices" for motion, the backend:
http://www.lavrsen.dk/foswiki/bin/view/Motion/WorkingDevices

As a follow up to your issues - I think that you should be able to use the RPi cam's you are testing - its just a matter of finding the correct settings.  I think that for these, you may need to use the "Remote MotionEye Camera" instead of network camera.  It looks like the example is misleading though (it suggests port 8080, etc.).

I found this issue where it is referenced, and they say that using port 80 with the remote motioneye camera is the fix:
http://www.lavrsen.dk/foswiki/bin/view/Motion/WorkingDevices.

So, give that a try if you'd like.  That might be your missing piece.  Here I believe is a screenshot of his working setup for adding a remote motioneye camear (it shows IP:80).
[image]

I would never have thought to use port 80 🤷 .

Cheers and good luck! I hope this is it!
DeadEnd

2boot, 2020-07-13:
@DeadEnded Thanks for that, I think I will succeed in getting the Pi to do as required, first with pizero, then I will move to the pi3b+ and then to sound. I will post further when I have something positive to say. It would be nice to be able to close this item on a positive basis.

Cheers!

2boot, 2021-04-09:
Out of my depth - closed

closed 2021-04-09 by 2boot