motioneye-project/motioneye #1114: Hikvison camera not fluid

SnakeDoc59, 2019-01-11:
Hello,

I'm sorry for my English, I'm french.
I just installed on an old Pi1 Raspbian with MotionEye application.

I implanted 2 of my 3 cameras on it.
1 Foscam FI8910w in wifi on my router
http://192.168.0.20:80/videostream.cgi

and 1 Hikvision in PoE on my switch
rtsp: //192.168.0.25: 554 / Streaming / Channels / 1

And the Pi1 is plugged into ethernet.

The cameras are well established but the image of the Hikvision is constantly frozen. I have the image, then frozen for 2/3 minutes, then the image etc ...

Do you have an idea why?
On VLC with the same flow on my pc I have no problem with fluidity.

Currently there is only foscam that permanently records on my OpenMediaVault server

Is the Pi1 not powerful enough?
In this case I'm not sure that with a Pi3 it will change much.

Thank you for your feedback.
Michael

SnakeDoc59, 2019-01-12:
Hello,

I think you did not understand my concerns. I have no problem with my server running in i7 with 12GB of ram, but with my raspberry.
I will try to run mytioneye with my cameras on a Pi3B + to see if it's more fluid

Chiny91, 2019-01-13:
My R Pi 3+ easily handles 3 cameras (all Hikvision).   I've previously used an R Pi 3 and (ISTR) an R Pi 2 for 3 cams.

Your Hikvision URL might need a tweak; maybe:
rtsp://192.168.0.25/Streaming/Channels/102 (for low res), or
rtsp://192.168.0.25/Streaming/Channels/101 (for high res), or
http://192.168.0.25/Streaming/channels/1/httpPreview (easiest on the R Pi).
Don't forget to config the Hikvision itself to select the res to be output by the cam.

It also helps to match motion to your camera resolution, or you will have your R Pi doing it, which is cpu intensive.   You can further reduce the strain on your R Pi by dropping the the frames per second; perhaps you have already done that.

Keeping "Captured before" to a low number (I use 2) is helpful.    Stop any hi-cpu processes and heavy bandwidth tasks, if you use your R Pi for non-motionEye tasks.

JamBalaya56562, 2026-11-09:
As Chiny91 answered, a Pi 1 is too slow to decode the main stream; the sub-stream (`/Streaming/Channels/102`) with the resolution in motionEye matching the camera's keeps it fluid. Closing, as this answers it.

closed 2026-11-09 by JamBalaya56562