motioneye-project/motioneye #1117: NTP freezes startup

peterevertz, 2019-01-14:
I setup a "pi zero w" with motioneyeos Version "motioneyeos-raspberrypi-dev20181210.img"
Everything was fine up to the moment I changed date syncronisation from "HTTP->google" to "NTP->my fritzbox"

Now the boot process stoppes at  "Setting current Date using NTP:" . Befor any ssh or login possibility.
I don't know why it does not work with fritz ntp server. At least a timeout is missing. 

I do not see another way as writing a new SD card, or resetting it on a different system.

michelecrudele, 2019-07-28:
I had the same failure on a Raspberry Pi3 with motioneyeos 20190427. 
Moving from HTTP Google to NTP ntp1.inrim.it and rebooting blocks any attempt to access the Pi3 and the camera server is not running. I have to flash the .img again
It didn't happen with one year ago version.

zagrim, 2019-07-28:
There's a separate github project for motionEyeOS, this might be more on topic there.

JamBalaya56562, 2026-11-08:
Closing, as zagrim pointed out above: setting the date at boot, over HTTP or NTP, is done by motionEyeOS, which is maintained in its own repository (motioneye-project/motioneyeos). Please report it there if a boot still hangs at "Setting current date using NTP".

closed 2026-11-08 by JamBalaya56562