motioneye-project/motioneye #1120: Pi Cam working but not showing on WebUI

almcbs, 2019-01-17:
Raspberry Pi 3 + Pi cam v2  = Camera 1
Raspberry Pi zero w + Pi cam v2 = Camera 2 
Both have instances of Raspbian with motioneye installed on top 
Monitoring and settings for both cameras are done via Camera 1 
Both cameras are working as expected and are detecting motion as videos are getting uploaded from both cameras

Issue: On the Web UI, I can only see the feed from Camera 2. Camera 1 shows as a camera with a slash through it. The settings for both cameras are very identical, with the exception of frame rate and rotation as Camera 2 is rotated vertically. 

I went through the "Raspberry Pi Camera Module Not Detected" troubleshooting guide, and all my settings and modules are in place. However, the command "raspistill -o cam.jpg" gives me the error below: 
pi@raspberrypi:~ $ raspistill -o cam.jpg
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

I get this error from both Camera 1 & 2 when I attempt to run that command. 
I also do not have the "special file" /dev/video0 present on either Cameras.

ccrisan, 2019-01-17:
@almcbs as far as I understand, you're using two separate boards for actual video capture but only one of them for actual viewing of the stream.

Please let us know:

 * if you can correctly view your Camera 2 stream by directly accessing the motionEye UI on the Camera 2 board
 * how exactly did you add Camera 2 to Camera 1 (Network Camera, Remote motionEye Camera, Simple MJPEG Camera)?

almcbs, 2019-01-17:
@ccrisan That is correct. 

- Yes. If I access Camera 2 motioneye ui I can view the Camera 2 stream. 

- Camera 2 was added as a remote motioneye camera.

almcbs, 2019-01-18:
This issue can be closed, I was able to resolve it. 

When I checked out the motion.conf file under directory "/etc/motioneye", there was really nothing to it. So I proceeded to copy over the motion.conf file from "/etc/motion" to the motioneye directory. Once I did that and rebooted, I was able to run the raspistill command (/dev/video0 still isn't present) successfully. When I logged into the WebUI, I was able to see the camera again as well. I made a copy of the thread-1.conf file prior to changing anything, so I just copied that back over to the motioneye directory and my camera was set up properly again.  

My next issue is Camera2 CPU running near 100% constantly, but ill open another issue for that in order to keep things from getting mixed up.

closed 2019-01-19 by ccrisan