motioneye-project/motioneye #1123: Start/stop a single camera (to reduce the total CPU load)

hsmptg, 2019-01-19:
I have a setup with 2 network cameras that create a CPU load of 80% in a Raspberry Pi 3.
I know that a possibility to start/stop all the server is using the meyectl (as suggested here: https://github.com/ccrisan/motioneye/issues/9).
But I would like to just start/stop a particular camera in order to reduce the total CPU load.
Is this possible?
Regards e thanks for this great project.

Artyum, 2019-01-21:
You can just disable video device for every cam independently

hsmptg, 2019-01-24:
Any help on this?
Thanks

duhuni, 2019-04-05:
This can be achieved by a python or shell script. Here are some pointers:

First, you have to indentify the config of your camera, it's one of the thread-*.conf files. On motioneyeos they are located in /data/etc/. 

```
[root@abc]# ls -la /data/etc/thread-*.conf
-rw-rw-rw-    1 root     root          2158 Apr  5 00:31 /data/etc/thread-1.conf
-rw-rw-rw-    1 root     root          1982 Apr  5 16:12 /data/etc/thread-3.conf
```

Now you remove the reference to that file from motion.conf. It looks like this. You remove the respective `thread thread-*.conf` line of the camera that you want to remove.

```
# @normal_password xxx
# @enabled on
# @show_advanced on
# @admin_username xxx
# @admin_password xxx
# @normal_username xxx

setup_mode off
webcontrol_localhost on
thread thread-1.conf
thread thread-3.conf
webcontrol_html_output on
webcontrol_port 7999
```

Finally, restart motioneye.

To restore the camera just add the reference back to the motion.conf and restart motioneye again.

JamBalaya56562, 2026-11-08:
Closing as answered above: switching a camera off in its settings stops Motion from reading it, which saves the CPU, and the comment above shows how to script that by removing the camera from Motion's configuration. Pausing only motion detection does not stop the decoding, so it saves less.

closed 2026-11-08 by JamBalaya56562