motioneye-project/motioneye #1125: Support for 1 wire in MotionEyeOs / Buildroot

cyb0man, 2019-01-20:
Really satisfied with the software, running the raspberry1 image on my Zero W and all is working well.
Started thinking i might want to add temp reading since this camera is sitting outside.

Found a really nice solution for this and I have a bunch of DS18B20 sensors laying around.
https://github.com/vvallee/motioneye-tools

The problem is now that I am not running Raspian (with 1 wire support), but MotionEyeOs instead which I really like!

Has someone solved this before me?

sdevrieze, 2019-11-02:
I also really would love support for DS18B20 temperature sensors: 
* overlay temperature in recorded images and videos
* record min/max temperatures per day, per week, per month, and annually
* min/max temperature thresholds setting: number of times a temperature went above or below a certain threshold per day, per week, per month, and annually

JamBalaya56562, 2026-11-13:
1-wire support is a matter of the OS image (kernel and device tree), not of motionEye: on Raspberry Pi OS it is switched on with `raspi-config` → *Interface Options* → *1-Wire*, and motionEye runs there as well, as suggested above and in #774. Showing the temperature in the overlay is #738. Closing here.

closed 2026-11-13 by JamBalaya56562