motioneye-project/motioneye #1129: Feature Request: Implement the Google Home Camera Guide

clogwog, 2019-01-23:
would it be possible to implement the Google Home interface so i can see the cameras via the Google assistant and/or Google Home Hub as well ?

The interface is described here:
https://developers.google.com/actions/smarthome/guides/camera
and it looks like you have to implement a response to a Sync command and supply a stream (which is already available in motioneye anyway)

ccrisan, 2019-01-23:
Not a planned feature, but I'd be happy to review a PR.

michaelwoods, 2019-01-25:
The stream would have to be in one of the following protocols (ie, not mjpg):
- HLS (HTTP Live Streaming)
- DASH (Dynamic Adaptive Streaming over HTTP)
- Smooth Streaming
- Progressive MP4 (most likely to be used only for Clips)

https://developers.google.com/actions/smarthome/traits/camerastream

NederWeb, 2019-09-18:
I would love to have this feature as well +1

SvenCelin, 2020-01-02:
This would be great feature! Is there any progress with this?

DewGew, 2020-01-21:
Looking for this also ..

crz16, 2020-02-08:
Would be awesome!

night4me, 2020-02-14:
via HA works fine with all protocols including mjpg, but would consider motioneye had this as native feature.

ilikecheesybread, 2020-05-25:
If that was a thing, I would definitely use it.

alexserb720, 2020-05-26:
it would be awesome

ursimon, 2020-05-28:
this would be amazing

leofuscaldi, 2020-09-03:
Would love that!

5b7p260, 2021-01-08:
it would be awesome

woutervddn, 2021-01-20:
I believe this should already be possible.
As described by [Home Assistant Documentation](https://www.home-assistant.io/integrations/nest/#camera) You can view nest/google camera's as an RTSP stream.

> All Google Nest Cam models, Google Nest Hello Video Doorbell, Google Nest Hub Max expose a CameraLiveStream via the SDM API, which returns a RTSP live stream which can be viewed from Home Assistant.

If we then look at the [Google Documentation for CameraLiveStream](https://developers.google.com/nest/device-access/traits/device/camera-live-stream), you can see how you can generate an RTSP stream.

In either case, I'm gonna try to get this working tonight. Probably as a standalone node program (as that is what I know best). I'll post my findings, an addon for this would be nice!

Dararii, 2021-02-07:
Is there any progress? I would love to use that feature.

DewGew, 2023-08-03:
Any progress on this request?

rezzalex, 2023-08-10:
Hello, 

Very interested as well.

Any news ?