motioneye-project/motioneye #1131: FNC and then no web interface?

SpyderMe, 2019-01-26:
I have a Pi Zero and had no issues out of the gate until I enabled FNC.  I seem to longer be able to access the web interface at all.  I realize some settings are not available once FNC is enabled, yet I don't see anywhere where the web interface no longer works.  I must be missing something.

The camera feed is coming though great on 8081.

ccrisan, 2019-01-26:
Your issue is related to [motionEyeOS](https://github.com/ccrisan/motioneyeos), not motionEye. What version are you running? Did you mess around with ports?

JamBalaya56562, 2026-11-08:
Closing, as ccrisan's questions above went unanswered. Fast Network Camera is a motionEyeOS mode, and motionEyeOS is maintained in its own repository (motioneye-project/motioneyeos), so please ask there if it still happens.

closed 2026-11-08 by JamBalaya56562