motioneye-project/motioneye #1132: motioneye _ file storage problem

jartobec, 2019-01-26:
Hi
I installed motioneyes under raspberry pi that works well except file transfer (file storage)

FTP with filezilla works to a personal website

The ftp with my freebox worked previously too, but since (update freebox !! ??) no transfer files

the test works (Test Service)
[image]

an extraction of the journalctl

janv. 25 09:59:09 raspberrypi meyectl[306]:    ERROR: failed to upload file "/var/lib/motioneye/Camera1/2019-01-25/09-59-01.jpg" with service ftp:
janv. 25 09:59:09 raspberrypi meyectl[306]: Traceback (most recent call last):
janv. 25 09:59:09 raspberrypi meyectl[306]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 875, in upload_media_file
janv. 25 09:59:09 raspberrypi meyectl[306]:     service.upload_file(target_dir, filename)
janv. 25 09:59:09 raspberrypi meyectl[306]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 98, in upload_file
janv. 25 09:59:09 raspberrypi meyectl[306]:     self.upload_data(rel_filename, mime_type, data)
janv. 25 09:59:09 raspberrypi meyectl[306]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 649, in upload_data
janv. 25 09:59:09 raspberrypi meyectl[306]:     conn = self._get_conn()
janv. 25 09:59:09 raspberrypi meyectl[306]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 685, in _get_conn
janv. 25 09:59:09 raspberrypi meyectl[306]:     self._conn.connect(self._server, port=self._port)
janv. 25 09:59:09 raspberrypi meyectl[306]:   File "/usr/lib/python2.7/ftplib.py", line 138, in connect
janv. 25 09:59:09 raspberrypi meyectl[306]:     self.welcome = self.getresp()
janv. 25 09:59:09 raspberrypi meyectl[306]:   File "/usr/lib/python2.7/ftplib.py", line 215, in getresp
janv. 25 09:59:09 raspberrypi meyectl[306]:     resp = self.getmultiline()
janv. 25 09:59:09 raspberrypi meyectl[306]:   File "/usr/lib/python2.7/ftplib.py", line 201, in getmultiline
janv. 25 09:59:09 raspberrypi meyectl[306]:     line = self.getline()
janv. 25 09:59:09 raspberrypi meyectl[306]:   File "/usr/lib/python2.7/ftplib.py", line 191, in getline
janv. 25 09:59:09 raspberrypi meyectl[306]:     if not line: raise EOFError

Thanks for advance

jm

ccrisan, 2019-01-26:
Can you try this with a different FTP server and see if the problem persists? It might not be a problem with motionEye but with your FTP server.

jartobec, 2019-01-27:
Hi
`Can you try this with a different FTP server and see if the problem persists?`
I tested different FTP server = result OK

`It might not be a problem with motionEye but with your FTP server.`

I looked for info on the FTP server  of my service providers to realize that I had two revisions of delay.

firmware version 4.0.1 = no problem with motioneye
December 2018  / firmware version 4.0.2 = problem with motioneye despite several reboot
?????????          / firmware version 4.0.3 = ??????????????
January 16, 2019 / firmware version 4.04 = **smooth operation with motioneye**(apply this day)

Murphy Law (December 2018 ......... January 2019)  :(
There is no notification for an update !!!!!!!! 

A big thank you for your help :+1: 
Kenavo
Bests regards

closed 2020-11-29 by jartobec