motioneye-project/motioneye #1133: Command to wake up Display

Neroson, 2019-01-26:
Hi,
I installed MotionEye on Raspbian and everything good so far.
The Raspberry Pi 3 B+ is connected to the offical touchscreen.
Now I want that the touchscreen turns off after a while and back on when a motion is detected.
I wrote a little script:

`#!/bin/bash`
`xset dpms force on`

I can get it work by manually executing the script, but not with "Run a Command".
In the log File it says: 
Xset: unable to open display

I think its because it execute the command by the user root.
I tried different things with no luck. 

I am not very experienced. Is there a solution or even better way?

Thanks for your efforts
Best regards

discarn8, 2019-01-26:
Read up on .Xauthority

https://stackoverflow.com/questions/11015029/cron-job-how-to-run-a-script-that-requires-to-open-display

JamBalaya56562, 2026-11-09:
As pointed out, the command runs as root outside the desktop session, so the script has to name the display and its authority file, e.g. `DISPLAY=:0 XAUTHORITY=/home/pi/.Xauthority xset dpms force on`. Closing, as this answers it.

closed 2026-11-09 by JamBalaya56562