motioneye-project/motioneye #1134: invalid credentials changing the URL

ssjboss, 2019-01-28:
Hello, 
I need to serve motioneye on the route /motioneye (http://myip:8765/motioneye/) so i have made some changes on the files:
Setting.py -> new variable base_url = '/motioneye'
server.py -> adding '+settings.BASE_URL+' on all lines on handler_mapping  
example:
r'^**'+settings.BASE_URL+'**/$ , handlers.MainHandler) 

On previous versions (0.21) this works and i can access to web url via http://myip:8765/motioneye/
but on the lastest version i can not login, because i says "invalid credentials"

Some help to make that working? thanks.

damianromanowski, 2019-02-04:
You plan on re-writing all those changes every release?

Why not just serve this behind a reverse proxy?

ssjboss, 2019-02-05:
fixed and working thanks

closed 2019-02-05 by ssjboss