motioneye-project/motioneye #1135: High CPU Utilisation since a few days

loxK, 2019-01-30:
Without any change, motion is using 100% all the time, whereas it used to to at 20%30%. 
My setup has 3 rstp camera. 

In order to fix it I upgraded motioneye to latest version and disabled motion detection. Video streaming is off as it always been. Video and image recording are motion triggered only.

Why can I do to debug that insane cpu usage ?

Motion 4.2.1
ffmpeg 3.4.1-1~16.04.york0

fngx, 2019-01-30:
Just an insight,

This might be posted in https://github.com/Motion-Project/motion as well, as you're using the rtsp capable version.

JamBalaya56562, 2026-11-10:
No details came after this, and with RTSP cameras most of the CPU goes to Motion decoding the streams, so this is more likely Motion 4.2.1 than motionEye. Lowering the frame rate or resolution, or turning on "Motion optimization" (as in #1161), usually helps. Closing as stale; please open a new issue with the logs if it happens on current motionEye.

closed 2026-11-10 by JamBalaya56562