motioneye-project/motioneye #1138: Future request

toshibochan, 2019-02-04:
Is possible to add option to use PIR senso instead of motion on video?

ccrisan, 2019-02-04:
Not yet.

toshibochan, 2019-02-04:
@ccrisan thanks for great work. Be great if can add PIR sensor. I made video doorbell using raspberry and Linphone python. I love to add motion sensor recording. But i don’t want use video motion because the trees moving in the wind. 
Thanks

jasaw, 2019-02-04:
@toshibochan A python script that monitors the GPIO state and triggers recording will do the job. Something like this: https://github.com/ccrisan/motioneyeos/issues/842#issuecomment-414375686

cbaymac, 2020-06-14:
I also would like to use PIR to trigger due to moving trees and wind. I’m not able to figure out how to activate the camera for a still or video with a python script. Any help would be appreciated. I should probably be using motion rather than motioneye. Having said that, I still need the python script to trigger the photos or video. Thanks.

numbatboy, 2021-04-02:
Ditto. Option for external trigger from PIR or camera's  facial recognition would be soooo useful. Also, would reduce processor load.

JamBalaya56562, 2026-11-13:
Since motionEye 0.45.0b1 (#2691), a PIR script, or anything else watching a GPIO pin, can start and end a motion event: `POST /action/<camera id>/eventstart` and `eventend`, after logging in once (`curl -c /tmp/me -d username=admin -d password=... http://<host>:8765/login/`, then `curl -b /tmp/me -X POST http://<host>:8765/action/1/eventstart`). With movies on, that records a movie.

closed 2026-11-13 by JamBalaya56562