motioneye-project/motioneye #1139: No camera on install Raspbian Stretch Pi 3B+ and Pi 3A+

joland, 2019-02-05:
hi,
I had 3 pi's running with motion eye os. 2 Zero W's and 1 Pi 3A+, having the Pi 3A+ as the Camera Hub and the 2 Zero W's connected to the Pi 3A+ through network camera. All was working perfectly. However I need certain linux features which the motion eye os does not support. So it would be great if I can have motion eye os running on stretch.

I follow the instructions multiple times on a fresh Raspbian Stretch install and updated it before starting the steps mentioned here: [https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian](url)

But in any case I am not able to connect to the camera (v2.1). I have tested on the Pi3B+ and Pi3A+.
What am i doing wrong?

ccrisan, 2019-02-05:
> But in any case I am not able to connect to the camera (v2.1).

Please provide more details. What error do you get? Screenshots often help.

joland, 2019-02-06:
The boot.log:
`[^[[0;32m  OK  ^[[0m] Started triggerhappy global hotkey daemon.
[^[[0;32m  OK  ^[[0m] Started System Logging Service.
[^[[0;32m  OK  ^[[0m] Started Disable WiFi if country not set.
[^[[0;1;31mFAILED^[[0m] Failed to start dhcpcd on all interfaces.
See 'systemctl status dhcpcd.service' for details.
[^[[0;32m  OK  ^[[0m] Started Save/Restore Sound Card State.
[^[[0;32m  OK  ^[[0m] Started Avahi mDNS/DNS-SD Stack.
[^[[0;32m  OK  ^[[0m] Started Login Service.
         Starting Load/Save RF Kill Switch Status...
[^[[0;32m  OK  ^[[0m] Started Load/Save RF Kill Switch Status.
[^[[0;32m  OK  ^[[0m] Started LSB: Autogenerate and use a swap file.
[^[[0;32m  OK  ^[[0m] Started LSB: Switch to ondemand cpu governor (unless shift key is pressed).
         Starting Load/Save RF Kill Switch Status...
[^[[0;32m  OK  ^[[0m] Started Configure Bluetooth Modems connected by UART.
         Starting Bluetooth service...
[^[[0;32m  OK  ^[[0m] Started Load/Save RF Kill Switch Status.
[^[[0;32m  OK  ^[[0m] Started Bluetooth service.
[^[[0;32m  OK  ^[[0m] Started Raspberry Pi bluetooth helper.
[^[[0;32m  OK  ^[[0m] Reached target Bluetooth.
         Starting Hostname Service...
[^[[0;32m  OK  ^[[0m] Started Hostname Service.
[^[[0;32m  OK  ^[[0m] Started Raise network interfaces.
[^[[0;32m  OK  ^[[0m] Reached target Network.
         Starting OpenBSD Secure Shell server...
         Starting Permit User Sessions...
         Starting /etc/rc.local Compatibility...
[^[[0;32m  OK  ^[[0m] Started Permit User Sessions.
My IP address is 192.168.1.200
[^[[0;32m  OK  ^[[0m] Started /etc/rc.local Compatibility.
         Starting Terminate Plymouth Boot Screen...
         Starting Hold until boot process finishes up...[^[[0;32m  OK  ^[[0m] Started Show Plymouth Boot Screen.
[^[[0;32m  OK  ^[[0m] Reached target Encrypted Volumes.
[^[[0;32m  OK  ^[[0m] Reached target Paths.
[^[[0;32m  OK  ^[[0m] Started Forward Password Requests to Plymouth Directory Watch.
[^[[0;32m  OK  ^[[0m] Found device /dev/serial1.
[^[[0;32m  OK  ^[[0m] Found device /dev/disk/by-partuuid/24c639f8-01.
         Starting File System Check on /dev/disk/by-partuuid/24c639f8-01...
[^[[0;32m  OK  ^[[0m] Listening on Load/Save RF Kill Switch Status /dev/rfkill Watch.
[^[[0;32m  OK  ^[[0m] Reached target Sound Card.
[^[[0;32m  OK  ^[[0m] Started File System Check on /dev/disk/by-partuuid/24c639f8-01.
         Mounting /boot...
[^[[0;32m  OK  ^[[0m] Mounted /boot.
[^[[0;32m  OK  ^[[0m] Reached target Local File Systems.
         Starting Create Volatile Files and Directories...
         Starting Preprocess NFS configuration...
         Starting Set console font and keymap...
         Starting Raise network interfaces...
         Starting Tell Plymouth To Write Out Runtime Data...
[^[[0;32m  OK  ^[[0m] Started Preprocess NFS configuration.
[^[[0;32m  OK  ^[[0m] Started Set console font and keymap.
[^[[0;32m  OK  ^[[0m] Started Create Volatile Files and Directories.
[^[[0;32m  OK  ^[[0m] Started Tell Plymouth To Write Out Runtime Data.
         Starting Update UTMP about System Boot/Shutdown...
         Starting Network Time Synchronization...
[^[[0;32m  OK  ^[[0m] Reached target NFS client services.
[^[[0;32m  OK  ^[[0m] Reached target Remote File Systems (Pre).
[^[[0;32m  OK  ^[[0m] Reached target Remote File Systems.
[^[[0;32m  OK  ^[[0m] Started Update UTMP about System Boot/Shutdown.
[^[[0;32m  OK  ^[[0m] Found device /sys/subsystem/net/devices/wlan0.
[^[[0;32m  OK  ^[[0m] Started ifup for wlan0.
[^[[0;32m  OK  ^[[0m] Started Network Time Synchronization.
[^[[0;32m  OK  ^[[0m] Reached target System Initialization.
[^[[0;32m  OK  ^[[0m] Listening on triggerhappy.socket.
[^[[0;32m  OK  ^[[0m] Listening on Avahi mDNS/DNS-SD Stack Activation Socket.
[^[[0;32m  OK  ^[[0m] Started Daily Cleanup of Temporary Directories.
[^[[0;32m  OK  ^[[0m] Listening on D-Bus System Message Bus Socket.
[^[[0;32m  OK  ^[[0m] Reached target Sockets.
[^[[0;32m  OK  ^[[0m] Reached target Basic System.
         Starting LSB: Autogenerate and use a swap file...
[... 24 lines cut ...]
[^[[0;32m  OK  ^[[0m] Started Load/Save RF Kill Switch Status.
[^[[0;32m  OK  ^[[0m] Started LSB: Autogenerate and use a swap file.
[^[[0;32m  OK  ^[[0m] Started LSB: Switch to ondemand cpu governor (unless shift key is pressed).
         Starting Load/Save RF Kill Switch Status...
[^[[0;32m  OK  ^[[0m] Started Configure Bluetooth Modems connected by UART.
         Starting Bluetooth service...
[^[[0;32m  OK  ^[[0m] Started Load/Save RF Kill Switch Status.
[^[[0;32m  OK  ^[[0m] Started Bluetooth service.
[^[[0;32m  OK  ^[[0m] Started Raspberry Pi bluetooth helper.
[^[[0;32m  OK  ^[[0m] Reached target Bluetooth.
         Starting Hostname Service...
[^[[0;32m  OK  ^[[0m] Started Hostname Service.
[^[[0;32m  OK  ^[[0m] Started Raise network interfaces.
[^[[0;32m  OK  ^[[0m] Reached target Network.
         Starting OpenBSD Secure Shell server...
         Starting Permit User Sessions...
         Starting /etc/rc.local Compatibility...
[^[[0;32m  OK  ^[[0m] Started Permit User Sessions.
My IP address is 192.168.1.200
[^[[0;32m  OK  ^[[0m] Started /etc/rc.local Compatibility.
         Starting Terminate Plymouth Boot Screen...
         Starting Hold until boot process finishes up...`

And screenshot:
[image]

ccrisan, 2019-02-06:
@joland this is your Raspbian boot log. That has nothing to do with motionEye. Make sure you have your `/dev/video0` device there. If you don't have it, motionEye can't do anything about it.

joland, 2019-02-06:
/dev/video0 is not there. 
Based on your comment i did a quick search and found:
> sudo rpi-update # to make sure the v4L2 drive is available.
> sudo modprobe bcm2835-v4l2 # to load it and create /dev/video0 --> i received an error

> raspistill -w 2592 -h 1944 -o foo.jpg #returned that the camera wasn't turned on

This is easily overlooked as motioneye os no such thing is required. You might want to add that to your wiki Install-On-Raspbian wiki.

Once i turned on the camera in raspi-config, the sudo modprobe created the /dev/video0 and all is running as it should.

OT. can i add network camera's running native motion eye os with the one above? I'm having trouble connecting them, especially when in want files to be stored at an usb stick on the device above...?

ccrisan, 2019-02-06:
> This is easily overlooked as motioneye os no such thing is required. You might want to add that to your wiki Install-On-Raspbian wiki.

It's right there, at the top of the [Install On Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) page.

> OT. can i add network camera's running native motion eye os with the one above? I'm having trouble connecting them, especially when in want files to be stored at an usb stick on the device above...?

Not sure what you mean by that, but see [this](https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios).

closed 2019-02-06 by ccrisan