motioneye-project/motioneye #1142: USB camera not displaying image

malvar0929, 2019-02-08:
Hello Crisan. I love the work you have done with MotioneyeOS. I'm running it on 3 Raspberry PIs very reliably. The issue I'm having is with a Raspberry Pi running Raspbian + Motioneye. I need to run it this way because I need to run other Java based servers. The problem is I add the USB camera, Logitech C615, there is no video displayed on the main web interface. However video is shown on the video stream interface although it looks like it's a very low frame rate but it works none the less. I have noticed that an older lower resolution Logitech where great but I need to use the higher resolution camera. Any ideas what could be causing this? Thank you in advance.

ccrisan, 2019-02-15:
> The problem is I add the USB camera, Logitech C615, there is no video displayed on the main web interface. However video is shown on the video stream interface...

It's not clear to me where you *can* and where you *cannot* see the video. What do you mean by "video stream interface"? Can you maybe attach some screenshots exemplifying the issue?

malvar0929, 2019-02-16:
Stream shows up on the stream port however no video is shown on the main interface.

[image]

ccrisan, 2019-02-16:
This is strange. Could you please open the developer console in your browser and watch for JavaScript errors? Attaching a screenshot of it should be good enough.

Amarok73, 2019-02-16:
Hello, 

It's the same effect as I describe in issue #1188 and still did not found the solution.

Amarok73, 2019-02-16:
@malvar0929 please execute command tail -f -n 50 /var/log/syslog then connect to the motioneye web interface and see if motioneye will start to flood the log with some specific messages...

malvar0929, 2019-02-17:
> @malvar0929 please execute command tail -f -n 50 /var/log/syslog then connect to the motioneye web interface and see if motioneye will start to flood the log with some specific messages...

Hello Amorok73
Please see messages from Motioneye. Hopefully this makes sense to you.
[image]
[image]

malvar0929, 2019-02-17:
Crisan please see Chorme console error. Thank you for your help
[image]

ccrisan, 2019-02-17:
@malvar0929 you appear to be having a monitor command that cannot be executed. Please remove any of your monitoring commands and see if the problem persists.

malvar0929, 2019-02-17:
> @malvar0929 please execute command tail -f -n 50 /var/log/syslog then connect to the motioneye web interface and see if motioneye will start to flood the log with some specific messages...

Please see images. Hopefully this makes more sense to you.
[image]
[image]

malvar0929, 2019-02-17:
> @malvar0929 you appear to be having a monitor command that cannot be executed. Please remove any of your monitoring commands and see if the problem persists.

This is a fresh install, all I did was add the camera. No monitor commands have been added by me. Where would I check to see if any have been inadvertently added?

ccrisan, 2019-02-17:
What files do you have in /etc/motioneye?

malvar0929, 2019-02-17:
You were right, there were some monitor entries in there. I deleted them and sure enough it fixed everything. Now that I recall, I think at some point I tried to restore a configuration from a previous installation of MotioneyeOS that I had running on this very RasPi and it must of added my old monitor entries. As always thank you for your great help and keep up the awesome work with Motioneye.

closed 2019-02-18 by ccrisan