motioneye-project/motioneye #1146: Camera Does not Require a Password

jethornton, 2019-02-13:
I have a D-Link DSC-5020L camera and it does not require a password. When I try and add it I get this error
cannot concatenate 'str' and 'NoneType' objects

jethornton, 2019-02-13:
Oh I forgot to mention it has a user name but no password

rotrials, 2019-02-14:
Please try something like: 

rtsp://ip:port/Streaming/Channels/101/
or
rtsp://username:password@ip:port/Streaming/Channels/101/

jethornton, 2019-02-14:
[image]
And how would that help the actual problem that motioneye tries to concatenate the empty password field?

ccrisan, 2019-02-14:
@jethornton please specify your motionEye version and please attach your [motioneye log](https://github.com/ccrisan/motioneye/wiki/Troubleshooting#getting-the-log-file).

jethornton, 2019-02-14:
motionEye Version | 0.39.3
Motion Version | 4.0
OS Version | Debian 9.7
[motioneye.log](https://github.com/ccrisan/motioneye/files/2866891/motioneye.log)

Thanks for the link to get the motioneye log.
JT

ccrisan, 2019-02-14:
@jethornton do you have your credentials embedded into the URL? If yes, can you please remove them and try again?

jethornton, 2019-02-15:
The URL is http://192.168.1.50/video.cgi?resolution=VGA and I put admin in the user box and nothing in the password box because it does not require a password.

ccrisan, 2019-02-15:
Can you open the `http://192.168.1.50/video.cgi?resolution=VGA` link with your browser and see a valid MJPEG? If yes, can I see a screenshot of it, just to make sure it's indeed a MJPEG stream without any surrounding HTML crap?

jethornton, 2019-02-15:
Here you go, I don't have any problem logging into the camera from a web page. The only problem is as above MotionEye fails because the password is empty.

[image]

ccrisan, 2019-02-15:
I have cameras with empty password and I can't seem to reproduce your error, that's my problem. The stack trace does not show up in your log file.

jethornton, 2019-02-15:
Does your camera need a user name but no password? That's what my camera needs. I don't have any problem if I supply no user name and no password or a user name and a password. If I put a user name and no password then the concatenation attempt fails with empty password field.

ccrisan, 2019-02-16:
@jethornton I configured one of my cameras to require user and no password, for this specific purpose. I can add it without problem.

Could you please set the [log_level](https://github.com/ccrisan/motioneye/wiki/Configuration-File#log_level) to `debug`, repeat the operation and see what the log has to say?

jethornton, 2019-02-17:
This is the only entry in the log after I attempt to add the camera and setting log_level to debug.
Feb 17 14:22:05 server meyectl[2412]: Failed to open /dev/video0: No such file or directory

JamBalaya56562, 2026-11-08:
Closing, as ccrisan could not reproduce it with a camera that has a user name and no password, and the debug log showed no traceback. This was the Python 2 release (0.39.3), where the error `cannot concatenate 'str' and 'NoneType' objects` came from. Please open a new issue with the debug log if a camera with an empty password still cannot be added with a current release.

closed 2026-11-08 by JamBalaya56562