motioneye-project/motioneye #1147: Network cameras on debian

joland, 2019-02-13:
Although, different scenarios have been described in [this](https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios) wiki. But how to actually set them up remains a struggle. 

I am trying to setup a camera network with motioneye on debian stretch. 

On the Zero W and 3A+ am i able to login with ip's:
http://192.168.1.200:8765 - vc.ril.camera - MMAL Camera - Camhub
http://192.168.1.201:8765 - vc.ril.camera - MMAL Camera - Cam1

Both camera's show footage and work as expected.

Trying to add Cam1 to the Camhub, i did the following steps:
- Select Network Camera
- URL: http://192.168.1.201:8765/
- add credentials

result: not a supported network camera
That is strange. On native motion eye os all worked without a hitch.

ccrisan, 2019-02-13:
Your camera is (probably) a Remote motionEye Camera, not just a regular network camera; so please select the correct camera type.

You could add it as a Network Camera, if you wanted, by using its streaming URL (http://192.168.1.201:8081, by default).

closed 2019-02-13 by ccrisan

joland, 2019-02-13:
By adding http://192.168.1.201:8081 as a Network Camera worked, but i get the same video output as Camhub.
If i add another http://192.168.1.202 as a Remote motionEye Camera, will be accepted. But does also output the same video output as Camhub.