motioneye-project/motioneye #1149: Feature request: download all videos/images

CosmonautX, 2019-02-14:
Hope this is the appropriate place for this!

Would it be possible to add a "download all" option in a future release? I'm generating around 1000-2000 photos a day and around 200-300 videos as my Pi+camera are pointing at a relatively busy area, and it would be useful to be able to simply download and archive these to a remote drive at the end of a day. The "zipped" option in photos doesn't work - a "failed to create zip" message appears every time - and with that number of files, it simply isn't practical to download them manually one at a time.

pundittech, 2019-05-19:
Yes please, I second this request. Thank you for your incredible work Sir.

ccrisan, 2019-06-03:
The zipped function fails due to too many files to be archived on a Raspberry Pi (limited CPU, RAM and card I/O). You have two options for this to work: either reduce the number of images by e.g. grouping them on an hourly basis or upgrade to a more powerful board (and SD card!).

Having zipped videos would fail the same way with fewer files given their relatively large size.

If you must download your files every day and you must have this many images per day, I'd recommend using SMB, FTP or SFTP to copy them to another server and remove them.

closed 2019-06-03 by ccrisan