motioneye-project/motioneye #1152: Banana pi pro wifi

Damianuscz, 2019-02-18:
Is it possible to use the internal WiFi? Add driver support?

ccrisan, 2019-02-18:
If this is about motionEyeOS, then no, there is no support for Banana Pi Pro.

Damianuscz, 2019-02-18:
This is a pity, because the version for banana pi works perfect. Only wifi do not work. So i must use usb wifi and can not use 2 usb webcam

ccrisan, 2019-02-18:
Are we talking about the Banana Pi M1 or the Banana Pro?

Damianuscz, 2019-02-18:
Banana pi m1 have same HW like banana pi pro. I have banana pi pro and run on it version of motioneye for banana pi m1. Banana pi pro have Internal WiFi http://linux-sunxi.org/Cubietruck/AP6210 but do not work and i must use usb wifi.

JamBalaya56562, 2026-11-10:
Drivers come with the OS image, so this belongs to motionEyeOS, and as said above it has no Banana Pi Pro build. Nothing to change in motionEye itself, closing. A request for the board would go to https://github.com/motioneye-project/motioneyeos.

closed 2026-11-10 by JamBalaya56562