motioneye-project/motioneye #1153: Missing features on Ubuntu?

Velkas, 2019-02-19:
I've got the latest version of MotionEye running on Ubuntu 18.04. Working nice and fast and have 2 MotionEyeOS enabled Fast Network Cameras running and connected. The problem I'm having is that it seems like my installation of MotionEye doesn't seem to have all the feature I expected given the features list and screenshots here in the wiki. I can select each camera I've added and I can change the associated name. That's about it. There's no recording, email alerts, motion tracking, or anything. Am I doing something wrong?

rotrials, 2019-02-19:
I assume you use motioneyeOS, please read: https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera

Velkas, 2019-02-20:
@rotrials So this isn't what I'm talking about. I have two Pi with cameras running MotionEyeOS specifically, as fast network cameras, because I'm running MotionEye (not OS) in Ubuntu on my computer as a hub server. The problem is that this hub server has no motion detection or recording in the UI. I can't figure out why.

rotrials, 2019-02-20:
@Velkas sorry, my mistake. What version of motioneye and motion do you use?

Velkas, 2019-02-21:
I'm running the latest of everything. I followed the instructions in the Wiki for Ubuntu.

motionEye Version | 0.39.3
-- | --
Motion Version | 4.1.1
OS Version | Ubuntu 18.10

Velkas, 2019-03-09:
@rotrials I apologize if this is inappropriate, just bumping this as I've seen no response.

rotrials, 2019-03-09:
@Velkas Sorry but I have no idea, is hard to say what is wrong. Is the "Advanced Settings" enabled? Is the motion, ffmpeg, v4l-utils installed ? (motion I think yes, because you provided the version). Did you get some error during the installation?

Can you provide some logs?

AZRadiohead, 2019-03-11:
@Velkas I just upgraded from 17.xx to 18.04 am and having issues.  I figured I'd try 18.10 and ran across your issue.  When you added your Pi cams, did you add them as MPEG cams or Network cameras with user and password?  I removed and re added mine the other day and noticed when I did added them as MPEG I had very limited options, like what you're describing.  I removed them and re added as Network cams, I think the second option and they came to life again in MotionEyeOS on my Ubuntu server.  I'll post any updates when I get 18.10 installed tomorrow.

Velkas, 2019-03-18:
@rotrials Thanks for trying at least. I appreciate the help.

@AZRadiohead Turns out, you're right. I've added them incorrectly to my hub. I re-added them correctly and all is working as expected!

closed 2019-03-18 by Velkas