motioneye-project/motioneye #1154: Green Screen from time to time

vplustute, 2019-02-20:
I installed the Motioneye on a Raspbery Pi 3B+ And connected 4 IP Cams with the Motioneye.
Three of them from the same Typ. From time to time I get a Green screen from this 3 IP Cams and after that I get an Motion detection.
What can be the problem?

TheBarret, 2019-02-26:
Isn't 4 cams on an RPI a lot of (perhaps too much) processing, might be due to lack of memory space or cpu power?

vplustute, 2019-02-26:
How can I find out if it is to much processing?
What is the maximum of cams what I can use with a Pi 3B+?

TheBarret, 2019-02-26:
I think that issue might just proof that :P

JamBalaya56562, 2026-11-10:
A green frame is a frame that could not be decoded, usually because packets of the camera stream were lost, so it also triggers motion. Using TCP for RTSP, a lower resolution or the camera's sub-stream usually fixes it, especially with four cameras on a Pi 3B+. No reply for years, so closing; please open a new issue if it still happens on current motionEye.

closed 2026-11-10 by JamBalaya56562