motioneye-project/motioneye #1155: Motion detection on night vision

vplustute, 2019-02-20:
The motion detection at daytime works fine. It took a long time to find the correct settings but it works. I don‘t know the funktion of „Dispeckle Filter“  and „Minimum Motion Frame“.
But the real problem is, that the motion detection at night vision Not realy works. All the settings I tried are not realy working.

TheBarret, 2019-02-26:
i experienced the same, i left "Dispeckle filter" off to get best results.

vplustute, 2019-02-26:
Ok thx, I will try it.

JamBalaya56562, 2026-11-10:
Night vision gives a noisy, low-contrast picture, so the same thresholds behave differently than by day. "Automatic Threshold Tuning" and "Automatic Noise Detection" adapt to that, and a mask for areas lit by IR reflections helps too. No reply since trying the hint above, so closing.

closed 2026-11-10 by JamBalaya56562