motioneye-project/motioneye #1156: pytz module missing from docker image

boombatower, 2019-02-21:
When using docker image and enabling timezone configuration in config: `#local_time_file /etc/localtime` the following is printed in log on boot.

```
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 11, in <module>
    load_entry_point('motioneye==0.39.3', 'console_scripts', 'meyectl')()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 255, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 439, in main
    run()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 362, in run
    make_media_folders()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 296, in make_media_folders
    config.get_main()  # just to have main config already loaded
  File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 185, in get_main
    _get_additional_config(main_config)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 2112, in _get_additional_config
    (sections, configs) = get_additional_structure(camera=bool(camera_id))
  File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 2083, in get_additional_structure
    result = func()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/tzctl.py", line 126, in timeZone
    import pytz
```

Assume all that is needed is installation of pytz module in `dockerfile`.

sreknob, 2019-04-12:
This appears to be fixed on the current development version.
Pull dev-amd64 or docker-amd64 tags for newer version.
Note you will have to start config from scratch or run the settings conversion script [migrateconf.sh](https://github.com/ccrisan/motioneye/blob/dev/motioneye/scripts/migrateconf.sh)

I downloaded the script and placed it in my config directory, then made executable and ran it without issue. Timezone appears to work correctly now.

`for file in /etc/motioneye/{motion,thread-*}.conf; do /etc/motioneye/migrateconf.sh $file; done`

JamBalaya56562, 2026-10-31:
This was still the case up to 0.44.0: with `local_time_file` set, the 0.44.0 image answers the web UI with 500 (`No module named 'pytz'`). The option and the time zone code using pytz were removed in #3466 (on `dev`, coming with v0.45.0): the line is now ignored with a warning, and the `edge` image works with it. To set the time zone of the container, mount `/etc/localtime` as shown in `docker/README.md`, or pass e.g. `-e TZ=Europe/Berlin`. Closing as completed.

closed 2026-10-31 by JamBalaya56562