motioneye-project/motioneye #1157: Question: Pass-through Recording

cbmboy, 2019-02-21:
I have been playing with Motioneye in Docker and it seems to work very nicely.  One thing I have been trying to understand is how to enable pass-through recording.  I don't see a clear label of this anywhere. Apologies, but I am very new to this, and don't know if I missed it.

Can someone point me to which setting to use to enable pass-through recording?  Do I just set the recording codec to the same as what is in my camera?  I expected this would instead decode, and then re-encode it back to the same - which is what I am trying to avoid.

Thanks in advance!

Quick follow up question - is it possible to add a sub-stream (record off main, view off sub-stream) without adding it as a second camera?  I didn't see a spot to do this, but again am not very experienced yet.

lmarceg, 2019-03-14:
I would also like to understand how pass-through is working. I tried to set up continuous recording, but I see the fps drop to around 2.4, which means the stream is decoded and then re-encoded (despite I selected H264/OMX, being my camera an H264 one).

bkiziuk, 2019-04-02:
You need to have the newest motion daemon (4.2) in order to have movie_passthrough option available.
As in motion 4.2 some command options have changed, you also need to have MotionEye 1.40 installed.
Then, either in MotionEye's camera config file or WebUI's `Extra Motion Options`, you need to put:
```
movie_passthrough on
```
parameter.
I use it like:
```
netcam_highres rtsp://192.168.0.8:554/live1.sdp
movie_passthrough on
```
and also use (this is actually Camera Device option in MotionEye):
```
netcam_url rtsp://192.168.0.8:554/live2.sdp

```
Therefore, I have low resolution for preview and high resolution for recording.

roger-, 2019-07-20:
I'm interested in using this to bypass transcoding my MJPEG source to H264 movies.

Any idea if this will work? If I use `movie_passthrough on`, what format should I select in the motioneye front end to save movies (which I assume will be ignored for MJPEG .avi's)?

I saw a related issue [here](https://github.com/ccrisan/motioneye/issues/1051) and created one [here](https://github.com/ccrisan/motioneyeos/issues/1998) BTW.

zagrim, 2019-07-20:
As [Motion documentation](https://motion-project.github.io/motion_config.html#movie_passthrough) say:
> When using a RTSP, RTMP, mjpeg and some V4l2 cameras, create movie files of the motion with the packets obtained directly from the camera.

When using `movie_passthrough on`, I'd very much expect Motion to completely ignore whatever motionEye has set `movie_codec` to. So, just leave "Movie Format" at whatever it is and see how the CPU load changes after adding `movie_passthrough on`. A drop in the CPU load by "motion" process should be quite clearly noticeable.
You see, as there's currently no explicit support in motionEye for passthrough recording, the GUI doesn't know how to adjust when you add Extra Motion Options.

roger-, 2019-07-20:
I added that to the video device settings and rebooted but nothing changed. It's still making H264 movies.

zagrim, 2019-07-21:
I tested this, with a netcam producing H264 MP4, by setting `movie_passthrough on` and setting "Movie Format" in motionEye to MKV. Strangely enough (well, I might just not understand how Motion processes the video stream), I got motion debug movies in MKV format while the recorded movies are MP4 (as I would expect them to be due to movie passthrough).

So, @roger- do you mean that you get the recorded movies (not the motion debug movies, if you happen to have "Create Debug Media Files" enabled) in the format specified in motionEye GUI? That wouldn't match my test results.

roger-, 2019-07-22:
I just changed the format to MPEG4 (.avi) and now it produces .avi files with the FMP4 codec.

I don't have debug files enabled. Is there a way to make sure `movie_passthrough on` is getting into motion? I noticed it's not in the motion.cfg file even after adding it to the GUI and rebooting.

zagrim, 2019-07-22:
Take a look at the camera specific config file (thread-*.conf or camera-*.conf depending on the application versions), it should be there.

My camera seems to also allow having a MJPEG sub-stream, so I tried it out. I'm not quite sure how to interpret my results, though, in contrast to your findings.

Sure enough the resulting file extensions look suspicious, but now, when hooked up to the MPJEG stream from the camera, `ffprobe` at least tells me that the codec being used is always Motion JPEG regardless of the file extension the files get. For example, with "Movie Format" set to "H.264/OMX (*.mp4)" I sure get *.mp4 movies but the codec really is MJPEG, and I get the same result when "Movie Format" is set to "MPEG4 (*.avi)". However, the debug files do change their codec as per set in "Movie Format", which I think is rather irrelevant (even though it is confusing). I do think it is confusing, though, that the recorded movie file extension is off from the codec being used, but I somehow seem to remember seeing somewhere a bug report for that already. I tried to find it but couldn't find such at least from motionEye project... maybe it was reported for Motion (as it probably should anyway, since it is Motion which is producing the movie files basd on the config motionEye generates for it.

How did you exactly check the codec for those movie files of yours? Both `vlc` and `ffprobe` are in agreement with the files I have (*.mp4 files with codec "mjpeg" or "jpeg").

roger-, 2019-07-22:
Thanks, looks like it is indeed in the camera conf. 

Hmm interesting. I checked the codec using VLC and didn't see MJPEG. Sounds like it basically works for you so it must be something off at my end. I'll double check everything.

jasaw, 2019-07-23:
Guys, with bypass recording enabled, it will still record to whichever file format you choose, either mkv or avi, because both are just containers. The encoded video stream within the container will be exactly the same as received from the IP camera.

The easiest way to check whether bypass recording is enabled is to check the recorded video. With bypass recording, motion is **not** able edit the frames, i.e. won't have timestamps and camera name in the video.

zagrim, 2019-07-23:
@jasaw Thanks for clarifying that. What you say about the container matches my test results so far, but only regarding the debug movies, not the recorded camera stream which for whatever reason was saved as .mp4 regardless of the selected "Movie Format" in motionEye.

Here's an example, for this run I selected Flash movies as the movie format. `ffprobe` tells the .mp4 files are using mjpeg codec, which is matches the camera stream (debug files get recognised as flv quite right).
```
/mnt/project/motioneye-dev/Camera1/2019-07-23/06-06-20m.flv
/mnt/project/motioneye-dev/Camera1/2019-07-23/06-06-20.mp4
/mnt/project/motioneye-dev/Camera1/2019-07-23/06-06-40m.flv
/mnt/project/motioneye-dev/Camera1/2019-07-23/06-06-40.mp4
```
So, basically I get .mp4 movies all the time, is the container somehow picked from the original stream by Motion, or something like that since nothing in the config points to .mp4? The main stream of the camera is h264 .mp4 so maybe it wraps also the mjpeg coded stream in mp4? That would be technically absolutely sensible, but for users not really proficient in video internals rather confusing.

jasaw, 2019-07-23:
@zagrim Check your camera-1.conf file and see whether `movie_codec` is set. I would expect motion to honour that option when bypass recording is enabled, but I can't be sure unless I read the source code.

zagrim, 2019-07-23:
@jasaw In my camera-1.conf I had this for the test run matching the result in my previous comment, so "mp4" didn't get picked from there, at least:
```
movie_codec flv
movie_passthrough on
```

DeadEnded, 2020-01-06:
Both of the original requests have been answered.
using the extra options to add the below to items enables pass-through recording, and second stream recording (so you set the initial camera on low def, but the netcam below as high res).
```
netcam_highres rtsp://IP/blah/blah/blah
movie_passthrough on
```
Additionally, I believe the newest release of MotionEye has included the movie-passthrough in the UI.  So you don't need to have this as an extra option anymore.

So, everything was answered (a while ago) so I am closing this issue.

Thank you to all who helped!
Apologies for leaving this open for so long!

Cheers!
DeadEnd

closed 2020-01-06 by DeadEnded