motioneye-project/motioneye #1159: Running on FreeBSD 12.0

neckroRO, 2019-02-23:
Hi, someone who is running MotionEye on FreeBSD maybe can give me a hint.
I`m try to run MotionEye on a FreeBSD 12.0 but i have a error regarding v4l-utils:

# meyectl startserver -c /etc/motioneye/motioneye.conf
    INFO: hello! this is motionEye server 0.39.3
 WARNING: you have motion installed, but no v4l-utils
   but, 
===>   v4l-utils-1.6.3 is already installed

Thanks!

TheBarret, 2019-02-26:
Seems to me motioneye is looking in the wrong paths or the v4l is not in the right folder.

Freebsd is ofc a different beasty all together compared to the others ;)

phaethon, 2019-04-14:
I am running on FreeBSD since yesterday. I had to change path in relayevent.sh, and at least some of the functions in diskctl.py are written in a Linux specific way (e.g. I cannot choose disk). I am using MotionEye as a server with all cameras connected being IP based, so I have not needed v4l-utils until now, but would be interested to hear any news/resolution of the issue.

closed 2026-11-01 by MichaIng