motioneye-project/motioneye #1161: Threads problem

nopmop, 2019-02-25:
I'd like to raise awareness about a possible problem with threads which imho gives quite evident symptoms: it seems that giving more tasks to operate upon evens out the usage of CPU cores, while with fewer tasks 1 thread/pool becomes more CPU hungry.
Here's a small example:

**Test setup**: RPi3-B+, RPi cam V2, gpu_mem=128, raspbian stretch. 

**Configs**:
- `1600x1200 + 30 FPS + no streaming` = one core stuck at 100%, others kinda idle
- `1600x1200 + 30 FPS + streaming` = all cores fluctuate evenly
- `max res (custom) + 4 FPS (= n. of Cores) + no streaming` = all cores mostly idle
- `max res (custom) + FPS > 4 + no streaming` = constant spikes in all cores

#988 #1051 #1135 #1122 #1029

I could look at the code (maybe port to green threads / coroutines too). Any pointers? 
PS: does the issue belong to motion, not motioneye?

closed 2019-02-25 by nopmop

ccrisan, 2019-02-25:
@nopmop what is the process (are the processes) that consume(s) most of the CPU cycles when you run your CPU-hungry scenarios?

nopmop, 2019-02-25:
@ccrisan it's `motion`. I was too frustrated by the 100% CPU issue so I opted to use another software, but I can revert back & still help debugging this if you want.

ccrisan, 2019-02-26:
@nopmop if it's Motion, then you're better off opening this issue on [their project page](https://github.com/Motion-Project/motion), although I doubt there's anything they can do about this particular issue.

Motion does its processing in the CPU, with a few exceptions, so at higher resolutions and/or framerates it's expected to see high CPU usage.