motioneye-project/motioneye #1162: Web Hook Stops working

kalliej, 2019-02-26:
If I restart the MotionEye service the Web Hook works perfectly for a while and then it just stops until restarting service again. Any idea?
Cannot find a file motioneye.log
No output trying: sudo journalctl -u motioneye | grep hook

kalliej, 2019-02-26:
The same thing happens when using a curl script - it only works for a while and then stops. It is still working if you type the url into a web browser. The ftp part (ftp the files to the cloud) keeps on working though...

JamBalaya56562, 2026-11-10:
No logs came for this, so the cause was not found. Current motionEye sends each webhook from a separate process with a timeout, and logs a failed request with its HTTP error, so a hanging call cannot stop later ones. Closing as stale; please open a new issue with `journalctl -u motioneye` output if it happens on current motionEye.

closed 2026-11-10 by JamBalaya56562