motioneye-project/motioneye #1163: how do you activate motion detection?

deostroll, 2019-02-26:
I can't find a direct documentation to this. I followed steps in the wiki to setup motioneye front-end on my raspberry pi 3B. But the front-end doesn't have a place where I can configure motion detection settings. That accordion is completely absent. Many tutorials show similar steps, but, when I followed them, I can't see that option in my web page.

[image]

I am trying to record videos only based on motion. Camera is at the door, and, I want to record people walking or ringing the door bell, and, waiting near the door.

Must I do something in motion.conf to get that?

rotrials, 2019-02-26:
I assume you are using motioneyeOS.
See: https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera

On the other hand, this doesn't come without some disadvantages:
- no more motion detection
- no more motion notifications
- no more pictures or movies
- no more overlaid text (date/time, camera name)

deostroll, 2019-02-26:
No. I am not using motioneyeos.

ccrisan, 2019-02-26:
Enable *General Settings > Advanced Settings* and you'll find plenty of stuff to tweak.

closed 2019-02-26 by ccrisan

vinz486, 2021-01-28:
MotionEye as Home Assistant 64bit addon on Raspberry Pi 4 8gb: no configurations available, useless.

[image]

starbasessd, 2021-01-28:
All HomeAssistant issues need to be discussed with the HomeAssistant
Community & FRENCK who forked motionEye for HomeAssistant.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?