motioneye-project/motioneye #1165: Setting base_path /cams for nginx reverse proxy - not getting CSS for interface

davidwhenderson3, 2019-03-01:
I am having problems setting the base_path /cams variable so my Nginx Reverse Proxy will work correctly.  I get the basic html like issue #298 install on debian - no gui but have no CSS.  I have motionEye installed on Ubuntu 16.04 and nginx is running.

[image]

I put the "base_path /cams" in the /etc/motioneye/motioneye.conf and enabled "server_name motionEye" as the hostname by removing the # - see below - however there is no /cams folder. Do I need to make a cams folder somewhere?

# overrides the hostname (useful if motionEye runs behind a reverse proxy)
# enabled by removing hashtag
server_name motionEye
base_path /cams

I have this configured in my /etc/nginx/sites-available/default file
If I put location /cams { instead of just location / {   -- I get a 404 error

# Default server configuration
#
server {
	listen 80 default_server;
	listen [::]:80 default_server;
	root /var/www/html; 
	# Add index.php to the list if you are using PHP
	index index.html index.htm index.nginx-debian.html;

	server_name motionEye;
	location / {
		proxy_pass http://localhost:8765/;
        	proxy_read_timeout 120s;
        	access_log off;
		# First attempt to serve request as file, then
		# as directory, then fall back to displaying a 404.
		try_files $uri $uri/ =404;
	}

My full server name is ubu16.dwhenderson.tech but with the override in the motioneye.conf file it should be just "motionEye" I think. Everything else works great and this is an tremendous program you wrote!! I am a IT college student and I using motionEye with 10 Raspberry Pi 3 & Pi 0 W and 12 cameras to present as a project. I want (it is required for my project) the nginx as an added layer of security.
Thanks!

ccrisan, 2019-03-04:
See https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx

closed 2019-03-04 by ccrisan

Wonderbox2000, 2020-04-24:
Hi I used to content of https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx

`location /cams/ {
        proxy_pass http://192.168.1.7:8765/;
        proxy_read_timeout 120s;
        access_log off;
    }`

Of course I changed the ip adress, but I am still getting the same webpage without any graphics.
I dont really understand the thing to change about the base path, can you help me please ?

DooMFeaR, 2020-12-27:
Same. Following the guide step by step. No css!