motioneye-project/motioneye #1167: Motioneye fail to load after pip upgrade

TobiLinux, 2019-03-03:
Hello,
My name is Sebastian. First of all, thank you very much for this excellent software!
The issue: After a pip upgrade to the latest available version, motioneye fails to boot. I am an enthusiastic user but I lack the knowledge to be able to solve it on my own.
OS: Debian 9
Here the output of systemctl:

● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-03-03 14:42:21 -03; 6s ago
  Process: 7000 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 7000 (code=exited, status=1/FAILURE)

mar 03 14:42:21 debian systemd[1]: Started motionEye Server.
mar 03 14:42:21 debian meyectl[7000]: Traceback (most recent call last):
mar 03 14:42:21 debian meyectl[7000]:   File "/usr/local/bin/meyectl", line 6, in <module>
mar 03 14:42:21 debian meyectl[7000]:     from motioneye.meyectl import main
mar 03 14:42:21 debian meyectl[7000]:   File "/usr/local/lib/python3.5/dist-packages/motioneye/meyectl.py", line 28, in <module>
mar 03 14:42:21 debian meyectl[7000]:     import settings
mar 03 14:42:21 debian meyectl[7000]: ImportError: No module named 'settings'
mar 03 14:42:21 debian systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
mar 03 14:42:21 debian systemd[1]: motioneye.service: Unit entered failed state.
mar 03 14:42:21 debian systemd[1]: motioneye.service: Failed with result 'exit-code'.

Thank you very much for the help!

ccrisan, 2019-03-03:
Use python 2.7. Install everything using `pip2`.

closed 2019-03-03 by ccrisan

TobiLinux, 2019-03-04:
Thx Calin!
It works now!