motioneye-project/motioneye #1168: Motioneye 0.39.3 fail to startup - Ubuntu server 16.04LTS

jjeliebenberg, 2019-03-05:
Hi there
When starting motioneye on Ubuntu 16.04LTS 64 bit server, it fails to start and I get the following error in the syslog:
Started motionEye Server.
`Mar  5 16:34:59 servxxx meyectl[2192]: Traceback (most recent call last):
Mar  5 16:34:59 servxxx meyectl[2192]:   File "/usr/local/bin/meyectl", line 9, in <module>
Mar  5 16:34:59 servxxx meyectl[2192]:     load_entry_point('motioneye==0.39.3', 'console_scripts', 'meyectl')()
Mar  5 16:34:59 servxxx meyectl[2192]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 254, in main
Mar  5 16:34:59 servxxx meyectl[2192]:     import server
Mar  5 16:34:59 servxxx meyectl[2192]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 28, in <module>
Mar  5 16:34:59 servxxx meyectl[2192]:     from tornado.ioloop import IOLoop
Mar  5 16:34:59 servxxx meyectl[2192]:   File "/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py", line 67
Mar  5 16:34:59 servxxx meyectl[2192]:     def fileno(self) -> int:
Mar  5 16:34:59 servxxx meyectl[2192]:                      ^
Mar  5 16:34:59 servxxx meyectl[2192]: SyntaxError: invalid syntax
Mar  5 16:34:59 servxxx systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Mar  5 16:34:59 servxxx systemd[1]: motioneye.service: Unit entered failed state.
Mar  5 16:34:59 servxxx systemd[1]: motioneye.service: Failed with result 'exit-code'. `

Any ideas how I can solve or fix this?
Thanks

blackdot, 2019-03-06:
Same here. (16.04.6 64-bit)

> systemd[1]: Started motionEye Server.
meyectl[16412]: Traceback (most recent call last):
meyectl[16412]:   File "/usr/local/bin/meyectl", line 11, in <module>
meyectl[16412]:     sys.exit(main())
meyectl[16412]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 254, in main
meyectl[16412]:     import server
meyectl[16412]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 28, in <module>
meyectl[16412]:     from tornado.ioloop import IOLoop
meyectl[16412]:   File "/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py", line 67
meyectl[16412]:     def fileno(self) -> int:
meyectl[16412]:                      ^
meyectl[16412]: SyntaxError: invalid syntax
systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: motioneye.service: Unit entered failed state.
systemd[1]: motioneye.service: Failed with result 'exit-code'.

Any advice?
Thanks

blackdot, 2019-03-06:
Not sure how it happened, but I had "tornado 6.0.1" on my system.

I uninstalled it
`pip uninstall tornado`
and then installed it again
`pip install tornado`

And now I have "tornado 5.1.1", motioneye is running now well!

daniel-widrick, 2019-03-06:
Same issue as @blackdot uninstall/reinstall worked like a charm

ccrisan, 2019-03-07:
Tornado 6.x no longer supports Python 2, so the solution is to use Tornado 5.x.

closed 2019-03-07 by ccrisan

jjeliebenberg, 2019-03-10:
Uninstalled tornado 6.0.1, but when 'pip install tornado' then it install 6.0.1 again.  
What is the appropriate way to install tornado 5?
When trying 'pip install tornado-5.1.1' then get message 'No matching distribution found for tornado-5.1.1'

jjeliebenberg, 2019-03-10:
Correct syntax is:
pip install tornado==5.1.1

aihysp, 2019-04-23:
yep same issue 
pip install tornado==5.1.1

did the trick