motioneye-project/motioneye #1169: MotionEye 0.39.4 not sending picture in motion notification mail

mcgyver83, 2019-03-05:
Hi, 
I'm running docker motioneye on a raspberry pi2 B.
I tried both with the official image (with motion 4.1.1)  and updating docker file to use motion 4.2.2 but I still receive notification mails without picture despite of I activated "Attached Pictures Time Span".

Any idea?

danch99, 2020-04-25:
Hi,
I have exactly the same problem.
Have you figured out a fix ?

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye 0.39.4 in Docker, and got no logs. The attached pictures are the still images the camera saved within the time span around the event, so they need Still Images on with motion triggered capture, and none are attached when none were saved in that span. Please open a new issue with `motioneye.log` if pictures are still missing with a current image.

closed 2026-11-08 by JamBalaya56562