motioneye-project/motioneye #1174: "Segmentation fault " error on raspbian stretch

Mybrc91, 2019-03-09:
```bash
sudo systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sat 2019-03-09 22:03:08 CST; 8min ago
  Process: 19311 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=killed, signal=SEGV)
 Main PID: 19311 (code=killed, signal=SEGV)
      CPU: 586ms

Mar 09 22:03:08 raspberrypi systemd[1]: motioneye.service: Unit entered failed state.
Mar 09 22:03:08 raspberrypi systemd[1]: motioneye.service: Failed with result 'signal'.
Mar 09 22:03:08 raspberrypi systemd[1]: motioneye.service: Service hold-off time over, scheduling restart.
Mar 09 22:03:08 raspberrypi systemd[1]: Stopped motionEye Server.
Mar 09 22:03:08 raspberrypi systemd[1]: motioneye.service: Start request repeated too quickly.
Mar 09 22:03:08 raspberrypi systemd[1]: Failed to start motionEye Server.
Mar 09 22:03:08 raspberrypi systemd[1]: motioneye.service: Unit entered failed state.
Mar 09 22:03:08 raspberrypi systemd[1]: motioneye.service: Failed with result 'signal'.
```

all motion log is blow
```bash [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motion/motion.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0:motion] [NTC] [STR] webu_start_strm: Starting all camera streams on port 8181
[0:motion] [NTC] [STR] webu_strm_ntc: Started camera 0 stream on port 8181
[0:motion] [NTC] [STR] webu_start_ctrl: Starting webcontrol on port 8180
[0:motion] [NTC] [STR] webu_start_ctrl: Started webcontrol on port 8180
[0:motion] [NTC] [ENC] ffmpeg_global_init: ffmpeg libavcodec version 57.64.101 libavformat version 57.56.101
[0:motion] [NTC] [ALL] translate_init: Language: English
[0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 0 is from /etc/motion/motion.conf
[0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 0 Camera Name: (null) Device: /dev/video0
[0:motion] [NTC] [ALL] main: Waiting for threads to finish, pid: 18993
[1:ml1] [NTC] [ALL] motion_init: Camera 0 started: motion detection Enabled
[1:ml1] [NTC] [VID] vid_start: Opening MMAL cam
[1:ml1] [NTC] [VID] mmalcam_start: MMAL Camera thread starting... for camera (vc.ril.camera) of 1280 x 720 at 15 fps
[1:ml1] [NTC] [VID] create_camera_component: MMAL camera component created
[1:ml1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[1:ml1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 4 items
```
Please help me, thank you.

ccrisan, 2019-03-10:
Please try running motionEye manually (using `meyectl startserver -d -c /etc/motioneye/motioneye.conf`) and see what process and at what point crashes. I suspect that Motion crashes and that's usually due to some ffmpeg issues.

Mybrc91, 2019-03-11:
> Please try running motionEye manually (using `meyectl startserver -d -c /etc/motioneye/motioneye.conf`) and see what process and at what point crashes. I suspect that Motion crashes and that's usually due to some ffmpeg issues.

Only print Segmentation fault, motion process not crash.

JamBalaya56562, 2026-11-08:
Closing, as there were no further replies. As ccrisan suspected, a segmentation fault here usually means Motion crashed, often in ffmpeg; this was Motion 4.2.2 with motionEye 0.40rc1. If Motion still crashes with current releases, please report it to [Motion](https://github.com/Motion-Project/motion) with its log.

closed 2026-11-08 by JamBalaya56562