motioneye-project/motioneye #1175: [0.40rc1] An error occurred. Refreshing is recommended... on Apply button

bob-lee, 2019-03-09:
Found a few previous issues but I am raising a new one here.

[image]

Steps:
1. From fresh installation of stretch lite on pi zero w, installed `0.39.3` with `motion 4.1.1`
2. Added MMAL camera with no further config, web shows camera ok
3. Uninstalled `0.39.3`, installed `0.40rc1` and ran `migrateconf.sh`
4. Reboot and web shows camera ok
5. Configured some, clicked Apply button, then got error

The changes seemed to have been saved ok as they appeared ok after the screen refresh.
Another problem is that I can't record a movie anymore, it keeps recording images even though Still Images sets to off and Movies set to on..?

Log:
```
pi@raspberrypi:/etc/motioneye $ sudo journalctl -u motioneye
-- Logs begin at Fri 2016-11-04 06:16:43 NZDT, end at Sun 2019-03-10 09:30:01 NZDT. --
Mar 09 23:37:59 raspberrypi systemd[1]: Started motionEye Server.
Mar 09 23:38:07 raspberrypi meyectl[242]:     INFO: hello! this is motionEye server 0.40rc1
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_interface"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_output"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "picture_quality"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_output_motion"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "picture_output"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_quality"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_codec"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "lightswitch_percent"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "text_scale"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "picture_output_motion"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_max_time"
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [NTC] [ALL] motion_startup: Logging to syslog
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar 09 23:38:17 raspberrypi motion[508]: [21706264:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar 09 23:38:18 raspberrypi meyectl[242]:     INFO: cleanup started
Mar 09 23:38:18 raspberrypi meyectl[242]:     INFO: wsswitch started
Mar 09 23:38:18 raspberrypi meyectl[242]:     INFO: tasks started
Mar 09 23:38:18 raspberrypi meyectl[242]:     INFO: mjpg client garbage collector started
Mar 09 23:38:18 raspberrypi meyectl[242]:     INFO: server started
Mar 10 09:21:12 raspberrypi meyectl[242]:  WARNING: 403 GET /login/?_=1552162885184&_username=_&_signature=75ee512e8d9f48d3231b780c09fbfa65f429c977 (192.168.1.73) 28.90ms
Mar 10 09:29:15 raspberrypi meyectl[242]:    ERROR: list indices must be integers, not unicode
Mar 10 09:29:15 raspberrypi meyectl[242]: Traceback (most recent call last):
Mar 10 09:29:15 raspberrypi meyectl[242]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
Mar 10 09:29:15 raspberrypi meyectl[242]:     raise_exc_info((type, value, traceback))
Mar 10 09:29:15 raspberrypi meyectl[242]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1756, in wrapper
Mar 10 09:29:15 raspberrypi meyectl[242]:     result = method(self, *args, **kwargs)
Mar 10 09:29:15 raspberrypi meyectl[242]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 290, in post
Mar 10 09:29:15 raspberrypi meyectl[242]:     self.set_config(camera_id)
Mar 10 09:29:15 raspberrypi meyectl[242]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 205, in wrapper
Mar 10 09:29:15 raspberrypi meyectl[242]:     return func(self, *args, **kwargs)
Mar 10 09:29:15 raspberrypi meyectl[242]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 522, in set_config
Mar 10 09:29:15 raspberrypi meyectl[242]:     set_camera_config(int(key), cfg, check_finished)
Mar 10 09:29:15 raspberrypi meyectl[242]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 371, in set_camera_config
Mar 10 09:29:15 raspberrypi meyectl[242]:     config.set_camera(camera_id, local_config)
Mar 10 09:29:15 raspberrypi meyectl[242]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 429, in set_camera
Mar 10 09:29:15 raspberrypi meyectl[242]:     adapt_config_directives(camera_config)
Mar 10 09:29:15 raspberrypi meyectl[242]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 158, in adapt_config_directives
Mar 10 09:29:15 raspberrypi meyectl[242]:     data.update(mapping(value, data))
Mar 10 09:29:15 raspberrypi meyectl[242]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 128, in <lambda>
Mar 10 09:29:15 raspberrypi meyectl[242]:     'text_double': lambda v, data: {'text_scale': [1, 2][v]},
Mar 10 09:29:15 raspberrypi meyectl[242]: TypeError: list indices must be integers, not unicode
Mar 10 09:29:15 raspberrypi meyectl[242]:    ERROR: 500 POST /config/0/set/?_=1552163367956&_username=admin&_signature=021e6694e9b829ef645b6f0eb922ae53e2aef10f (192.168.1.73) 99.82ms
pi@raspberrypi:/etc/motioneye $
```

retrostrobe, 2019-03-10:
I'm getting a the following error when trying to add a `Local MMAL camera: VideoCore camera`. I can however add a `Video V4L2 camera: mmal service 16.1` without any issues.  

```
Mar 10 12:57:43 rbpi3 meyectl[752]:     INFO: adding new mmal camera with id 2...
Mar 10 12:57:43 rbpi3 meyectl[752]:    ERROR: 'video_controls'
Mar 10 12:57:43 rbpi3 meyectl[752]: Traceback (most recent call last):
Mar 10 12:57:43 rbpi3 meyectl[752]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
Mar 10 12:57:43 rbpi3 meyectl[752]:     raise_exc_info((type, value, traceback))
Mar 10 12:57:43 rbpi3 meyectl[752]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1756, in wrapper
Mar 10 12:57:43 rbpi3 meyectl[752]:     result = method(self, *args, **kwargs)
Mar 10 12:57:43 rbpi3 meyectl[752]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 293, in post
Mar 10 12:57:43 rbpi3 meyectl[752]:     self.add_camera()
Mar 10 12:57:43 rbpi3 meyectl[752]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 205, in wrapper
Mar 10 12:57:43 rbpi3 meyectl[752]:     return func(self, *args, **kwargs)
Mar 10 12:57:43 rbpi3 meyectl[752]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 691, in add_camera
Mar 10 12:57:43 rbpi3 meyectl[752]:     camera_config = config.add_camera(device_details)
Mar 10 12:57:43 rbpi3 meyectl[752]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 567, in add_camera
Mar 10 12:57:43 rbpi3 meyectl[752]:     camera_config = motion_camera_ui_to_dict(motion_camera_dict_to_ui(camera_config), camera_config)
Mar 10 12:57:43 rbpi3 meyectl[752]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 809, in motion_camera_ui_to_dict
Mar 10 12:57:43 rbpi3 meyectl[752]:     vid_control_params = (('%s=%s' % (n, c['value'])) for n, c in ui['video_controls'].items())
Mar 10 12:57:43 rbpi3 meyectl[752]: KeyError: 'video_controls'
Mar 10 12:57:43 rbpi3 meyectl[752]:    ERROR: 500 POST /config/add/?_=1552175863599&_username=admin&_signature=f001c79e0309fd66ea5c7ecfbefd1b54b0a08d8d (10.0.1.6) 22.69ms
```

bob-lee, 2019-03-10:
@retrostrobe I had that issue when I installed `0.40rc` on fresh stretch image. Then I followed steps described in OP. Maybe related to this issue, not sure.

ccrisan, 2019-03-10:
Guys, indeed, the 4.1 - 4.2 options mapping was not working well for < 4.2. This should be fixed by 46857733481e2671e277626ffcc42796be3ad008 and present in `0.40rc2`.

Thanks for testing the new version and for reporting the problem.

bob-lee, 2019-03-10:
Great, uninstalled `0.40rc1`, installed `0.40rc2`, could Apply config changes, can now record movies. Closing the issue.

closed 2019-03-10 by bob-lee