motioneye-project/motioneye #1176: Permissions issue on saved movies

NginUS, 2019-03-11:
On Linux Mint 18.3, Firefox 65.0.1, Motion 4.2.2, MotionEye 0.39.3.

I have a symlink from /var/lib to /var/lib/motioneye/CameraName elsewhere which is owned by user.

When it records a movie, the /var/lib/motioneye/CameraName/Date folder (& its contents) are owned by root. This is less than ideal.

Is there a way to set these files to be owned by user?

NginUS, 2019-03-11:
After following the instructions on the wiki here:
https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User

I wound up re-configuring everything from scratch, which is fine- but even using the same configuration values the camera will not display its image. I get the grey placeholder every time.

Earlier when I had it all running as root, I wrote down all the values I used in each field to make it work. eg. Camera URL, Resolution, Frame Rates, etc. Using these values after running the service as user fail. I can log into the MotionEye Web UI, but adding the camera fails to show its image.

As soon as I revert to running the service as root it works fine. But I have the problem with its files saving as root again.

JamBalaya56562, 2026-11-08:
Closing, as the wiki steps for running as a normal user are no longer needed: since 0.43.1 the service that `motioneye_init` installs runs as the unprivileged `motion` user, which owns the media folders, so new recordings are not owned by root. The steps are in the [README](https://github.com/motioneye-project/motioneye#installation). Please open a new issue if the camera still shows no picture with a current release.

closed 2026-11-08 by JamBalaya56562