motioneye-project/motioneye #1177: [0.40rc2] An error occurred. Refreshing is recommended... on adding camera

bob-lee, 2019-03-11:
steps:
1. on `0.40.rc2`, remove camera, `camera-1.conf` file disappeared
2. add camera, got this error

log:
```
Mar 11 14:18:54 raspberrypi meyectl[210]:     INFO: removing camera config file /etc/motioneye/camera-1.conf...
Mar 11 14:19:20 raspberrypi meyectl[210]: Failed to open /dev/video0: No such file or directory
Mar 11 14:19:25 raspberrypi meyectl[210]:     INFO: adding new mmal camera with id 1...
Mar 11 14:19:25 raspberrypi meyectl[210]:    ERROR: 'video_controls'
Mar 11 14:19:25 raspberrypi meyectl[210]: Traceback (most recent call last):
Mar 11 14:19:25 raspberrypi meyectl[210]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
Mar 11 14:19:25 raspberrypi meyectl[210]:     raise_exc_info((type, value, traceback))
Mar 11 14:19:25 raspberrypi meyectl[210]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1756, in wrapper
Mar 11 14:19:25 raspberrypi meyectl[210]:     result = method(self, *args, **kwargs)
Mar 11 14:19:25 raspberrypi meyectl[210]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 293, in post
Mar 11 14:19:25 raspberrypi meyectl[210]:     self.add_camera()
Mar 11 14:19:25 raspberrypi meyectl[210]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 205, in wrapper
Mar 11 14:19:25 raspberrypi meyectl[210]:     return func(self, *args, **kwargs)
Mar 11 14:19:25 raspberrypi meyectl[210]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 691, in add_camera
Mar 11 14:19:25 raspberrypi meyectl[210]:     camera_config = config.add_camera(device_details)
Mar 11 14:19:25 raspberrypi meyectl[210]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 573, in add_camera
Mar 11 14:19:25 raspberrypi meyectl[210]:     camera_config = motion_camera_ui_to_dict(motion_camera_dict_to_ui(camera_config), camera_config)
Mar 11 14:19:25 raspberrypi meyectl[210]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 815, in motion_camera_ui_to_dict
Mar 11 14:19:25 raspberrypi meyectl[210]:     vid_control_params = (('%s=%s' % (n, c['value'])) for n, c in ui['video_controls'].items())
Mar 11 14:19:25 raspberrypi meyectl[210]: KeyError: 'video_controls'
Mar 11 14:19:25 raspberrypi meyectl[210]:    ERROR: 500 POST /config/add/?_=1552267178766&_username=admin&_signature=dfb0a4fa1124321da8245d03785c02e1d838751e (192.168.1.73) 101.34ms
```

Also tried to remove motion `4.1.1` then installed `4.2.2`, got the same error.

ccrisan, 2019-03-11:
@bob-lee what type of camera are you trying to add (v4l2, netcam, mmal, remote)?

bob-lee, 2019-03-11:
@ccrisan I am using mmal camera.

ccrisan, 2019-03-11:
Fixed in 443ab40c3608b645c2b0ba8e7d93c7d42b13d09b.

closed 2019-03-11 by ccrisan