motioneye-project/motioneye #1179: ERROR: the maximum number of tasks (100) has been reached ( on a RPI3)

mulleer, 2019-03-12:
Hi,
I am using a motionEyeOs on a RPI3 :
- wifi connection active
- standard RPI cam
- taking a picture every 15 seconds
- storing them on a USB key
- uploading them on my Google Drive (I've tried with DropBox too with the same issue)
After a while and some tens of pictures uploaded, the system remains active (streaming ok) but no pictures are stored any more and no new uploads.
In motioneye.log, I see plenty of repeating of messages (every 15 seconds)  "ERROR: the maximum number of tasks (100) has been reached....".
Thank you for your feedback !
Eric

ccrisan, 2019-03-12:
Long story short: too many media files are created. More media files are created than can be uploaded to Google Drive or Dropbox. You'll need to reduce them by adjusting your pictures and/or movies settings.

closed 2019-03-12 by ccrisan

mulleer, 2019-03-14:
[MotionEyeOs issue with Google Drive & DropBox.pdf](https://github.com/ccrisan/motioneye/files/2965514/MotionEyeOs.issue.with.Google.Drive.DropBox.pdf)

Hi Crisan,
The problem occurs only when specifying folders (see attached document).
Thank your you attention.
Eric

adx-git, 2021-05-29:
Sorry for bringing this old ticket up.
Will there be a fix? I run into the same bug.

starbasessd, 2021-05-29:
@adx-git 
https://github.com/ccrisan/motioneye/issues/1179#issuecomment-472169401