motioneye-project/motioneye #1180: Motion detection deactivated but still triggering alarms

dr400, 2019-03-14:
I've noticed two situations that might be linked : webcontrol action does not work. When motion detection is paused, it reverts back to active within a couple of seconds (60s approx.). In the meantime, when motion detection slide is on "off", system still detects motion and triggers emails, catches pictures and movies ... Sounds weird ...

dr400, 2019-03-14:
the only solution to really stop detection is to switch off "still images" and "movies" alomg with MD. Is this related to the automatic reversion from "pause" to "active" when webcontrol is used ?

dr400, 2019-03-15:
is the problem coming from the coexistence of motion and motioneye ?

dr400, 2019-03-15:
Anyone to help ?

zagrim, 2019-06-13:
Are you still having this issue? I thought I should try to test this as I need to anyway figure out the possibilities of pausing and resuming motion detection. So, I configured my motioenEye test setup with really sensitive MD and went on to pause MD using motion web control. It's been now 6 minutes since the last bytes were written on disk and MD stays paused, so everything seems to working ok in my setup.

You mention "motion detection slide is on "off", system still detects motion and triggers emails": In my case there's nothing in motionEye UI that seems to reflect that MD is paused other than a "-" shown in the camera stream in place of the number of changed pixels (as I have "Show changed pixels" on). Are you saying that in your case motionEye sliders change when you use the webcontrol actions? What URLs are you actually referring to (I might be in the dark on some of the options there are)?

And some further questions that would help figuring this out :)
Which version of motionEye (and motion) you're actually using (you can get the info by adding "/version" at the end of your motionEye URL)?
Does this occur even if you just watch motionEye screen without touching anything after pausing MD? Do you have any custom scripting that try to operate motion or motionEye (scheduled or not)?
Could you show us the logs around the time when the issue occurs?

zagrim, 2019-06-14:
Ah, @dr400, I think I figured out the reason for your issue. Do you have Working Schedule set? It will continuously enforce the status of `motion` based on the schedule, so whatever you do manually will be overridden soon (see #1240).
Your only option right now might be to implement some daemon on the side that would command `motion` via the web control actions in the way you require, or to disable Working schedule in motionEye.

Skandy, 2019-08-02:
Hi,

I have the exact same problem...

motionEye Version	0.40
Motion Version	4.2.2
OS Version	motionEyeOS 20190427 (on raspberrypi 2)

Does anyone know why motion detection is automatically restarted ?

Thanks.

https://groups.google.com/d/msg/motioneye/pxYcDo7BA5M/sx8rJd2jAQAJ

On Wednesday, June 12, 2019 at 9:49:26 AM UTC+2, M B wrote:
Hey Guys,

I use these commends too but after a few seconds the status changes again to ACTIVE and the detection is recording again.

Any idea how to fix that?

If i "pause" the detection i get an prompt: Camera 1 Detection paused.

If i "status" i get the answer: Camera 1 Detection status PAUSE

BUT some seconds later "status" again: Camera 1 Detection status ACTIVE

Thx 4 the Help

zagrim, 2019-08-03:
@Skandy , referring to my comment directly above, do you have Working Schedule set?

Skandy, 2019-08-03:
@zagrim the Working Schedule option is turned off.

JamBalaya56562, 2026-11-08:
Closing as fixed by #3468 and #3506. As zagrim found, the working schedule kept turning detection back on; besides, motionEye read the state back from Motion every 10 seconds and restored its own, which undid a pause sent straight to Motion's webcontrol even without a schedule. Since #3468 motionEye no longer polls that state, and since #3506 a pause from the detection button lasts until the schedule changes. Checked in #1240 with Motion 4.5.1: a webcontrol pause stayed paused after 45 seconds. Both are in 0.45.0b1 and will be in v0.45.0.

closed 2026-11-08 by JamBalaya56562