motioneye-project/motioneye #1182: Can we record remotely?

deostroll, 2019-03-15:
There are two machines. A raspberry Pi 3b running motioneye software, and a pi zero installed with motioneyeos. I added the 2nd camera to the motioneye UI (1st machine). Is there someway we can have the pi zero feed recorded and stored somewhere on the first machine?

victorsvedberg, 2021-01-25:
Did this ever get solved/Answered? I'm happy with the functionality of MotionEye, but I to want to record slave cameras onto a master/server, should be a pretty common request, to prevent criminals from just snatching the camera and walk away with the footage.

starbasessd, 2021-01-25:
You can create a motionEye or motionEyeOS Hub to do all the heavy lifting
(motion detection & notifications), and or upload pics and videos to FTP,
SMB/Windows shares, GDrive, Dropbox, etc. The functionality has always been
there.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-01-25 by starbasessd

victorsvedberg, 2021-02-12:
> You can create a motionEye or motionEyeOS Hub to do all the heavy lifting (motion detection & notifications), and or upload pics and videos to FTP, SMB/Windows shares, GDrive, Dropbox, etc. The functionality has always been there.
> […](#)

Maybe I've missed something but it seems to me that the recording needs to be done by a remote machine to ensure that the recorded material is stored securely. The "solution" you mention above seems to be flawed since the motion event will be uploaded after the motion event, so if the thief destroys the camera, the footage might never be uploaded.

starbasessd, 2021-02-12:
You set the Network Camera, or PI to stream.
You have another motionEye or motionEyeOS Hub to connect to the stream and
do the motion detection and recording.
For example, I have 5 cameras around, 3 d-Link DCS5020Ls, a FosCam 9900,
and a motionEye with an attached webcam.
They all record locally to themselves, and they all have streams I can
connect to.
I have another motionEye hub set up that connects to all 5 streams, and can
do its own motion detection and recording. The streams from the 5 devices
are all continuous, so the seperate hub can do it's own 'heavy lifting.
If the cameras were not recording, but only had the streams, up until the
camera getting stolen/disconnected, the hub (in another part of the
building), would still be recording and I'd have the video recordings there.
Let me know if you need further explaination...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

wecoon, 2022-05-20:
> You set the Network Camera, or PI to stream. You have another motionEye or motionEyeOS Hub to connect to the stream and do the motion detection and recording. For example, I have 5 cameras around, 3 d-Link DCS5020Ls, a FosCam 9900, and a motionEye with an attached webcam. They all record locally to themselves, and they all have streams I can connect to. I have another motionEye hub set up that connects to all 5 streams, and can do its own motion detection and recording. The streams from the 5 devices are all continuous, so the seperate hub can do it's own 'heavy lifting. If the cameras were not recording, but only had the streams, up until the camera getting stolen/disconnected, the hub (in another part of the building), would still be recording and I'd have the video recordings there. Let me know if you need further explaination...
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

@starbasessd I would like to get an additional explanation. So I've installed motioneye on rasbian. I have Pi Zero running camera, and Pi 4 running the hub. I've connected the camera to the hub, and I can see the stream, but options I can configure on the hub is actual camera options, not hub options. So, I can only enable recording on the camera device. How could I record on the hub?