motioneye-project/motioneye #1185: Save Videos To USB Drive From Remote Camera

TheModdersDen, 2019-03-16:
So I currently have a setup where I have a Raspberry Pi 3 b+ as a "master" raspbian os motioneye instance and a Raspberry Pi Zero W "slave" with the same software setup. I followed this tutorial to get a RAID 1 setup up and running: https://blog.alexellis.io/hardened-raspberry-pi-nas/ to save videos for an extended time alongside other personal files (creating a NAS and camera "hub", per say).

The issue is that I do not see my USB RAID drives as an option in the dropdown menu (picture attached and highlighted). I have also tried entering the mount path manually, but with no luck there as well.

Just a couple of questions that I have that may resolve it off the bat:
- Are there any special permissions that MotionEye needs to write to a particular directory?
- Does MotionEye support USB drives?
- If MotionEye does support USB drives, does it support RAID (specifically RAID 1)
- Can motioneye even take video from a "slave" camera and save it to a "master" raspberry pi? If so, how?

Hope this all made sense and I hope to hear back soon! 😅 

As promised, the picture:
[image]

TheModdersDen, 2019-03-16:
Thought I should mention this: it looks like I am thinking along the lines of the outline in issue #1182. The major difference is that I have my USB drives in a RAID system. 

So, while it's close, I do not think it is a duplicate (in my opinion).

starbasessd, 2021-01-25:
If you are still having issues, open a new ticket. Closing this one for being 2 years old and no traffic.

closed 2021-01-25 by starbasessd