motioneye-project/motioneye #1187: Latest Version - Still Getting Admin Password Change Issues

Saleshorse, 2019-03-17:
I hate to admit this happened to me but I just installed MotionEye after trying to set up something similar on Raspian. I saw and read through the related issues and none of them are helping. I was adjusting the advanced settings to set rotation, frame rate etc. 

- On Pi Zero W
- Applied advanced settings
- Chrome asked it should save the updated password, I agreed (though didn't change it)
- Chrome tells me the password is now *****
- Attempted old password, *****, SHA sums of each combination you previously listed (*****, null, "null", my old password, etc...).
- Cleared Cache, cookies, everything
- Tried Incognito 
- Tried Edge (yuck)
- Chrome for Android
- Another PC

I've iterated over all the combinations I can think of, nothing has helped. I even have a saved copy of the config I exported and it has the SHA of my intended password. I've tried all the same through SSH to no avail. It's actually running and running well, thus far I'm super excited about this setup (and have 3 more cameras to setup). I really don't want to rip the card and start over just from the Wifi issues I faced on the router side.

Saleshorse, 2019-03-17:
Don't ask why I tried it, but I used the password I provided for the remote site (SFTP) and this actually got me in with the admin user. 

The input from this line:

`'upload_password': $('#uploadPasswordEntry').val(),`

Actually updated the admin password. Google Chrome didn't catch it either, it also thought it was the "*****" that we did on a previous thread. Fun times. Also works through SSH.

Saleshorse, 2019-03-17:
Sorry, one more thing I noticed, Chrome detects the password change to ***** when I click into the photo browser as well. I had the admin panel up and clicked the scene, then photo browser and it asked if I wanted to save my updated password. It didn't persist, but just an FYI.

ccrisan, 2019-03-20:
@Saleshorse please try the latest 0.40rc5.

JamBalaya56562, 2026-11-10:
The cause was Chrome autofilling the settings form, so the upload password landed in the admin password. Autofill was turned off for the settings in 0.40 (8cb4e46e), and every password field is `autocomplete="new-password"` now. Closing as fixed.

closed 2026-11-10 by JamBalaya56562